kwindrem/GuiMods #1: Feature: Time to full charge ?

akileos, 2021-04-19:
Hello,
 Many thanks for the excellent work, you're far ahead of my own tinkering and I switched to using yours instead of mine.
One thing I couldn't figure and that I miss a bit is how to add a guesstimate how remaining time to full charge using the missing Ah as a reference.

Thanks again !
-RB

kwindrem, 2021-05-08:
I feel that any guesstimate I could make on remaining charge time would be wildly inaccurate. Charging efficiency is part of the issue. Absorption time is the big issue, taking between minutes and hours for the battery to reach 100% after switching out of bulk mode.

akileos, 2021-05-08:
Totally makes sense, was thinking about missing Ah / current charge rate but didn't account for the different charge phases and thus was totally wrong. Good to close 👍 Thanks again !

closed 2021-05-08 by akileos