kwindrem/GuiMods #2: Hide Tanks option?

davemuench, 2021-04-26:
First off, amazing work, Victron needs to pick this up and merge it.

Is there a combination of setup options that would get the enhanced flow/views on my Touch GX 50 but not show the tanks? Thanks!

kwindrem, 2021-04-26:
The setup script has separate enables for each piece:
Flow enhancements
  Power gauges under that
Enhanced Mobile Overview
  this automatically includes the tank enhancements
  The tanks column is always present and empty if you have no tanks.
  I agree it's a waste of space but that's the way it is currently with or without my mods
Move Settings and Notifications to the top of the list
Hide the tile overview

davemuench, 2021-04-26:
Thanks, no problem - hopefully I will add some tank sensors sooner or later.

closed 2021-04-26 by davemuench