kwindrem/GuiMods #3: White screen when enabling "Enhanced Mobile Overview Page"

CzarofAK, 2021-05-06:
I have played around with your mod. it is very promissing.
sadly i encountered the issue, of a white screen instead of the "Enhanced Mobile Overview Page". The page with the tile is still present.
Firmware installed: v2.70 5 large 18
- SignalK = on
- Node-Red = on

kwindrem, 2021-05-06:
This package can now be installed on v2.70~5-large-18.

The problem is that the file set was incomplete AND there were bugs in SetupHelper that allowed an install when the file set was incomplete.

You need to grab the latest copy of GuiMods and SetupHelper, then do a manual install.

GuiMods is unique in that there is a version-depending file but there is no original, so the logic that attempts to create a new file set on a working system has no reference for which replacement file to put in the new file set. Working on a solution but nothing comes to mind yet.

CzarofAK, 2021-05-07:
Did work for me. Great work, thanks

closed 2021-05-07 by CzarofAK