kwindrem/GuiMods #4: Tanks Tile doesn't respond to touches

rmaddy, 2021-05-07:
I just installed your GuiMods. During install (on a 2.66 system) I said yes to all of the changes except for the power gauges. I like the change. Nice job.

Once I scroll to the enhanced mobile overview screen, the whole right third of the screen is the Tanks tile. I have no tanks setup so this whole tile is blank except for the title and the "No tanks found" message.

If I try to swipe left or right to bring back the hub overview, nothing happens if the swipe begins anywhere within the Tanks tile. The bottom toolbar with the Pages and Menu options also won't appear (or disappear) if the screen it tapped anywhere within the Tanks tile.

kwindrem, 2021-05-08:
That is because the tank column will scroll if the tiles get too small.

So I made a change so that scrolling is enabled only if it's needed to access all tiles.

This should help you with your page scrolling and bringing up the bar at the bottom of the page

kwindrem, 2021-05-08:
I have also just added support for the compact tiles introduced in v2.70~9 which will mean that the column won't need to scroll until more than 10 tanks are displayed (13 if there's no pump button).

rmaddy, 2021-05-08:
Thanks. I've since been making some tweaks of my own based on your enhanced mobile overview. Since I don't have any tanks connected to the RPi I have removed that whole tile. I'm going to get this looking just like I want to under 2.66 just in time for 2.70 to change it all. :)

On an unrelated note, I figured out how to show the time in "h:mm ap" on the Settings screen. It's the file /opt/victronenergy/gui/qt-components/qml/Qt/labs/components/native/StatusBarInternal2.qml.

The change is pretty obvious once you look at the file. Would like me to open another issue with the details or is this comment enough (assuming it's something you want to add or not)?

kwindrem, 2021-05-08:
What you are doing may make sense for many users that don't run Venus on a boat or RV. So I would suggest you post your changes in your own GitHub account.

One of the additions others have done is to add temperature readings to the tanks column. This makes a lot of sense because that area of the Mobile Overview not typically all used.

Nice catch on the time of day. I'd made changes on the Mobile Overview and "Flow Overview" pages. I missed the Device List title bar. You should post that change also.

So far, the changes for v2.70 are limited to the tanks column and hiding the pump button. (I hid the pump button on my enhanced layout too.)

The mechanism I use to make modifications saves an original version along with each modified file. From these, it's easy to identify changes between Venus OS versions and to make new replacements for versions as they are released. I've put some effort into the mechanism and you are welcome to use it. Check out the readme for SetupHelper.