kwindrem/GuiMods #5: Backup of default GUI

5teveO, 2021-05-13:
Not an issue as such but i'm helping someone that knows nothing about SSH etc to update to your GUI and they have asked if there is any way to back up the original GUI in case 'anything goes wrong' I've updated my install to your GUI and appreciate the work you have done.. the person I am helping is a little nervous about the process.. 

Steve

kwindrem, 2021-05-13:
The setup script backs up any files that get modified or replaced. Running setup again and choosing (u) uninstall will put everything back to the way it was prior to the install.

I'd be open to suggestions on how to make the installation process easier and less worrisome.

I have some scripts I run on my Mac to update my Venus devices with my mods, but these are unix shell scripts and run as a command line call so not sure that makes things any better. I originally had setup scripts that would run on a unix host to download an initiate the install/uninstall but abandoned this because it only worked from a Mac or unix machine and was complicating my setup scripts.

rsync works well to transfer files from a unix host to the Venus device. It preserves symbolic links where scp does not.

5teveO, 2021-05-13:
Hi Kevin

Great thank you .. I must have missed that due to me doing SSH on my phone.. :)

I dont think it is worrysome or anything like that. I guess less technical persons that rely on other to do certain areas of works are always a bit hesitant to change things.. so just need a bit of reassurance.. 

Keep up the great work.. 

Steve