kwindrem/GuiMods #6: Victron Venus OS GUI

jefndeb, 2021-05-26:
Hello, I installed your Venus GUI mod and it is awesome!!!! I was wondering, (i am not much of a coder though), is there a simply way instead of having the 2 fuel tank levels on the main screen replace them with my 2 temp sensors? (fridge and cabin temp)
Thanks, Jeff

kwindrem, 2021-05-26:
Working on adding temp sensors to the list now. As long as the sensors show up in the devices page they'll show up in the "tanks" column when I'm done.

kwindrem, 2021-05-26:
Temperature sensors have been added to the "tanks" column. Any temperature sensor that appears as a dBus services should appear.

Tile heights adjust to show as many tanks and temperature sensors as practical. Tiles "squeeze" to make room for more, then they "compact" to show even more. In compact mode, all information appears on a single line.

After compacting tiles, if there are still more than will fit on the screen, the tanks and temps columns become "flickable" -- you can scroll the list vertically to see additional tanks or temperature sensors.

The temperature range displayed in the bar is from -30 °C to +60 °C. With 0 in the middle, the bar grows out of the middle either towards the left or the right. Negative temps have a bluish tint, positive temps have a greenish tint.

Since there is no temperature setting in Venus, I display temps in both Centigrade and Fahrenheit.

Let me know if you have any suggestions for improvement.

BTW, I believe you can turn off tank sensors for Venus devices with inputs for them. My Venus device is a Raspberry PI, so there are no analog tank inputs available so can't help with turning them on/off.

jefndeb, 2021-05-27:
wow, super cool, I guess I have to reload the software?....(i actually didnt do it myself, I had a software friend remote into my laptop and using the putty app, he uploaded the new GUI....i watched but cant say I followed it 100%.
Jeff

kwindrem, 2021-05-27:
Yes, you need to download the new version of GuiMods and run the setup script. The ReadMe has instructions to update GuiMods, but in the end you need to run the setup script to reinstall.

kwindrem, 2021-05-27:
I found what I believe is a reasonable way to add temps to the row at the bottom.

For 4  or less tanks + temps exist, a single row of items is shown
For more items, the row is split into separate rows for temps and tanks
Each row scrolls horizontally if more than 4 items exist

jefndeb, 2021-05-27:
ok, thats what I thought I saw when the GUI was first loaded (2 fuel at the bottom) but I dont currently have any tanks connected so I turned those off, then I thought maybe I can change those to refirg temp and cabin temp.....just gotta figure out how...

kwindrem, 2021-05-27:
Depends on your Venus device. I believe Cerbo has some analog inputs that can be used for temperature sensors. I don't have any so I'm guessing about how to enable them. But once you've got them showing up, you can set the sensor's name and that will appear on the overviews.

There are only three tank types: battery, Fridge and generic so you don't have much to choose from. That only affects the color of the tile so you can use what's there for limited differentiation between sensors. Have fun.

jefndeb, 2021-06-01:
So, if my Venus (Cerbo) is connected to the internet, as it is, to update to have the 2 temp sensors on the main page, I just type this command into the SSH session?

wget -qO - https://github.com/kwindrem/GuiMods/archive/current.tar.gz | tar -xzf - -C /data
mv /data/GuiMods-current /data/GuiMods

jefndeb, 2021-06-01:
[image]

kwindrem, 2021-06-01:
After the wget and mv commands:

/data/GuiMods/setup

then follow the instructions and answer questions and enter necessary data.