kwindrem/GuiMods #7: Reinstall after Venus update

5teveO, 2021-05-26:
Hi

I have just updated Venus and lost the mod.. apart from running setup again, is this meant to reinstall itself after an update?

I'm not on the latest version.. it's a few weeks old.. 

Steve

kwindrem, 2021-05-26:
If you updated via the on-line or USB stick mechanisms, then it should have reinstalled automatically. /var/log/SetupHelper will most likely contain the reason an update failed. 

You can run setup manually and choose reinstall.

There are conditions that prevent auto reinstall however. The most common is setup not being able to create the necessary replacement files. I have made this mechanism more robust in the last couple of weeks so you might want to get a new copy of GuiMods and SetupHelper then attempt a manual install.

5teveO, 2021-05-28:
So I'm not too good at this kinda thing..

I have just been to the boat and tried downloading and installing again. It keeps telling me there is an incomplete data set for 2.70 15... Won't run the MV command as it says directory isn't empty, so I deleted it and now it says the directory doesn't exist.. so I guess I have stuffed things up..

Could you advise how I can remove the GUI mods manually or via a script.. and start again please.. I only have SSH access on the boat as it's a 4g connection and haven't managed to get a VPN working yet...

Steve

5teveO, 2021-05-28:
Thanks Kevin..
No apology required, you do this and share for free and this is probably my screw up anyway! I'm grateful for the work and support you have given. 

Yes boat has 4g connectivity (which in itself is causing me issues with remote access via VPN) so will run the details you gave and report back. Really appreciate the help..

Steve

5teveO, 2021-05-29:
OK Reporting back

Thanks that worked incredibly well and now loving the new enhancements.. great work.. and thanks so much again!

Steve