kwindrem/GuiMods #11: GUI Refinements

neoneddy, 2021-08-11:
I've been working on some refinements. Reducing the font size on some of the details and trying to get power gauges to fit into the overall design better.  (changed main color to white, made 70% transparent).
[image]

BY any chance do you know why the pane crashes when I change the height of the power gauges?

Overall I'm just trying to bring some visual hierarchy to it all, so it's easier to scan.  I think as it is now all the information kind of screams and everything is crashing (or getting darn close).  I hope I'm not offending here.

kwindrem, 2021-08-11:
The multicolored backgrounds on the gauges is an indicator of how close you are to the next level of caution. Without them, you won't know you are a sliver below the yellow zone. I wouldn't want to give that up.

Space is tight on this screen and I tried to reach a compromise between readability and presented information. If you have a big screen, smaller fonts wouldn't be a problem but on say a 3.5" or 5" screen, smaller fonts might not be visible.

I have no idea why increasing gauge height would cause the code to crash. Maybe there's clue in the log. There are calculations for the bar height within the overall gauge height and if you make the bar too high, some calculations will go negative. But that shouldn't be a problem if the overall height increases.

neoneddy, 2021-08-11:
> The multicolored backgrounds on the gauges is an indicator of how close you are to the next level of caution. Without them, you won't know you are a sliver below the yellow zone. I wouldn't want to give that up.

I agree the functionality  and information is great.  I'm exploring some other ideas on how to keep it informational but also fit into the overall design.   I've been thinking more about the gauge over the Multiplus's red/orange stripe.  It's a great place for it, but I'm sure VIctron would like to keep that bit of branding.   

> Space is tight on this screen and I tried to reach a compromise between readability and presented information. If you have a big screen, smaller fonts wouldn't be a problem but on say a 3.5" or 5" screen, smaller fonts might not be visible.

Another great point, I'm on a 7"  higher DPI screen so I'm seeing more space than others might have.   I think this is the same size as the Venus GX screen.  I'd say it should be readable on the CCGX .  If an rPi user is using a smaller screen I think they should stick with the more information dense overview. 

I appreciate your insight and help, this is your labor of love and it's never lost on me.

kwindrem, 2021-08-11:
It might be possible to orient the gauges vertically along the outside of each tile. But you run into space for systems where 5-6 digits are required for power. One way around this is to switch from watts to kilowatts when the power exceeds 9,000. The other issue with the edge is the rounded tiles. 

Gauge on the side won't work for the Multi tile though as there are flow connectors on both sides and horizontal space is limited. You could displace time and locate that elsewhere but the only space is below the multi and I moved ESS state there.

Space is especially tight with tanks and temps at the bottom of the screen. Without them, you can make most tiles larger.

neoneddy, 2021-08-11:
I'm kicking this idea around.  I took a screenshot from the stock flow view and applied the gauges.  Any omitted data is not intentional.  Use a rounded rectangle with a white border and bars inside it.  Then use a single black bar set at 75-80% opacity to adjust in size anchored to the top.  This should reveal the gauge bars and feel more like stacked LEDs lighting up. 

On the battery side, I'm not sure exactly how to handle it.  in fact I'm not sure yet how this gauge works.  It appears it is a gauge that goes positive and negative?  I'm in an RV so at times my alternator does push 100+ amps into the battery and I'll draw 100+ amps.  I figure, could do a custom gauge where it goes up or down or has tow black overlays anchored top and bottom?  Or two gauges, one for charge, other discharge?

I know we're losing pixels and resolution on the gauges, but I think knowing where you're at on capacity for inverting / shower power / generator etc is still pretty a parent.

I'm also thinking about a button for the settings on the AC IN side, keep the whole panel clickable, but I discovered it by accident.  Might help with usability.

[image]

kwindrem, 2021-08-11:
The gauges don't need a lot of resolution but the DO need to show the ranges reltive to the current value. That is: where the current inverter power is relative to it's maximum continuous limit. Another project I worked on had a narrow bar next to the main one to show the limits.

Also consider that some systems have two or three phases and each one needs it's own bar. My current bars split into two/three in that case.

I started the flow overview project because I wanted more information at a glance and many were asking for voltage and current. 

I think the key to doing this correctly is going back to LESS information on the tiles. What information is really necessary as an overview? I think the gauge is more useful than even the main power number, but others may disagree. Voltage and frequency are really used to diagnose issues: e.g., why is my shore power being rejected? I'm not sure if total power or individual phase powers are more important. I can see the answer depending on the system or even the user.

In any case, simplifying the tiles will require a way to get to more detail quickly is essential. That seems to be the consensus from people using the stock overview: it takes too long to dig through the menu system for details.

I though about using a tap on the tile to "zoom in" to more detail much like I did with the inverter mode and input current limit pop-ups. However, swiping to other overviews and to call up the bottom menu buttons can't be made more difficult and if every tile taps to zoom in, there's little real estate to swipe! So maybe a "details" button on each tile would work. Maybe that can be worked into the tile's banner.

Another reason to REMOVE information from the Flow overview is it's slowing down. On my Cerbo, it takes 2 seconds to switch from the menu to the overview!

The battery gauge does show both charge and discharge with zero being somewhere in  between. In my system, I can discharge twice as fast as charge, so the zero point is at the 2/3 point. Showing positives and negatives on the same gauge may require rotating the gauge horizontally again as vertical space is often limited. My existing AC input and Multi tiles do not show negative values, however they should: a Multi charges or discharges and with feed-in, AC input could also be negative also.

In another project, we discussed the gauge switching between one that shows positive numbers and one that shows negative numbers, but that can be jarring if you are close to the zero point.

neoneddy, 2022-02-10:
@kwindrem Coming back to this, going to do some hacking on my own install.   In the end, I love the work you're doing, I'm just trying to get it to look a bit more on brand.  I mean no offense, I love the functionality of it, I'm just trying to apply some branding and style guide type stuff to it.