kwindrem/GuiMods #14: Errors? Binding loop detected for property "width"

DevoDave, 2021-08-29:
Looking at the logs I see a whole bunch of what I assume are errors..

`@40000000612b1bf32ebf9144 file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:153:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1bf32ebfbc3c file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:138:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1bf32ebfcfc4 file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:138:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1bf42fc09df4 file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:153:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1bf42fc2c4bc file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:138:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1bf42fc93cfc file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:138:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1c1f2feb8e24 file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:153:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1c1f2febb91c file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:138:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1c1f2febc8bc file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:138:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1c2030c16c24 file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:153:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1c2030c19334 file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:138:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1c2030c1a6bc file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:138:5: QML Rectangle: Binding loop detected for property "width"
@40000000612b1c6b353c5584 file:///opt/victronenergy/gui/qml/PowerGaugeBattery.qml:153:5: QML Rectangle: Binding loop detected for property "width"
`

kwindrem, 2021-08-29:
Yes, these are from my code and I have no idea how to fix it. They don't seem to be preventing the code from running. 

Any ideas?

DevoDave, 2021-08-29:
I'm not even at the 'Hello World' coder level for this QML stuff yet so this is mainly guesswork but I offer the following. Maybe a fresh eye even if it has no idea somehow provide a clue.

In both those error lines (#138 and #153) width is being set by calling function barWidth().

barWidth() sets the color of the bar depending on charge current, and sets barcolor (which I guess is global in scope) and returns either (current x scalefactor) **OR** (-current * scalefactor). Does whacking a minus in front of a variable name actually convert that negative value positive value? Would (**-1 * current**) be better?  Is there an ABS() equivalent in this QML stuff?

kwindrem, 2021-08-30:
Looked over the code and found the problem. The calculations were all correct however a situation could have caused width to be recalculated when nothing had changed. I found another mechanism to trigger the calculations.

There were binding loop warnings in PowerGauge.qml also that got fixed also. The new version v2.13 includes the fix.

Rikkert-RS, 2023-04-30:
Hello kwindrem,

I also had the same problems with the PowerGauge.qml with the latest version. The log is constantly flooded with errors, also not good for SD cards.
I looked at the code and found a solution. The loop is created by the "barXoffset" because no "root.barXoffset" is used here.

I changed all 
bar1offset to root.bar1offset
bar2offset to root.bar2offset
bar3offset to root.bar3offset

in the Rectangle bar1 to bar3 and the functions calculateBar1width to calculateBar3width

the Erro QML Rectangle: Binding loop detected for property "width" no longer appears.

best ragards Rikkert-RS

kwindrem, 2023-04-30:
Thanks. I'll make these changes in the next release which should be out in the next day or so.