kwindrem/GuiMods #15: MultiPlus controls on Enhanced Mobile Overview report a failure

CzarofAK, 2021-09-12:
Hello @kwindrem, great modes, pretty impressive and  very easy to use! thanks a lot.

i have no expierinced, that if i enable the "Enhanced Mobile Overview" the Multiplus controls are disabled ("AC CURRENT LIMIT" and "AC MODE")
See failure message below:
[image]

see my Victron Community issue:
https://community.victronenergy.com/questions/101447/multiplus-does-not-allow-changes-via-venus-os.html

kwindrem, 2021-09-12:
I suspect the issue is unrelated to GuiMods. Uninstall my GuiMods and check the default Mobile Overview to see if it has the same problem.

The alert being shown indicates that the system thinks there is more than one inverter/charger in the system so the current limit and AC Mode can't be adjusted from the Mobile Overview.

CzarofAK, 2021-09-12:
When i enable your option, the two function don't work anymore. if i disable the option, its back on. I spent the whole day on this, and have to confirm, that it is really down to that single option.

CzarofAK, 2021-09-12:
I have only one Multiplus, a Lead-Acid Battery, a SmartShunt and a GX Tank 140 connected to my test setup

CzarofAK, 2021-09-12:
If i can do anything to help you, let me know.

kwindrem, 2021-09-12:
I've made some changes to the test for too many inverters. Grab a new copy and see if that fixes your issue.

If not, I need to more about your system
What components do you have connected to your GX device?

What firmware version are you running?

If you can, provide a list of dbus services reported by dbus-spy.

CzarofAK, 2021-09-13:
pretty impressed by your update function. guimods on 2.15 is working well. thanks a lot!

closed 2021-09-13 by CzarofAK