kwindrem/GuiMods #16: change preset value for screen backlight?

virtuvas, 2021-09-21:
on my setup the approx 1/5 of full brightness that seems to be the preset value is way too dark.
After every reboot I have to go there and up the value. Unfortunately the new value doesn't survive a reboot so I have to do it again and again.
is there any easy way to edit the preset value? or someway to save the brightness value as edited on the screen?

cheers

V.

kwindrem, 2021-09-22:
There is a bug in Venus OS that incorrectly sets the maximum brightness on some systems. 

The slider in the Display menu uses the max value from /sys, not the dbus setting parameter so it works correctly. But the value stored in settings is used during boot-up to restore the previous brightness. That value is limited incorrectly.

This is addressed in my RpiDisplaySetup repo:

`        dbus -y com.victronenergy.settings /Settings AddSettings '%[{"path": "/Gui/Brightness", "default":255, "min":0, "max":255}]' > /dev/null
`

Make sure the max value is correct for your display. E.g., for the Touch 50/70 the max is 15.

virtuvas, 2021-09-22:
thanks,

using  dbus-spy
com.victronenergy.settings /Settings/Gui/Brightness reports a value of 100, and fits with the position on the scroller to real max  being 255.

run the dbus command above, now works fine!

V.

closed 2021-09-22 by virtuvas