kwindrem/GuiMods #18: USB Mk3 VE Bus inteface for  Inverters stops power bars and status on overview screen

RobDuthie, 2021-09-28:
Just testing more features of the Gui mods.
When using  the USB to VE bus Mk3 inteface it kills the power bars and status on the inverter.
When you turn off the power bars and the enhanced flow overview it stops the status showing on the inverter blank bars.
When you do reboot before the Mk3 finds the inverter the power bars and status are there for a few seconds then are wiped clean.
When using a USB  VE direct  inteface, inverter power bars are OK with enhanced flow overview. But when you turn them off the inverter status is blank again.
Tested on 2 separate raspi 3B+ units with own inverters and interfaces running versions on 2.80.19 and  upgraded to 2.80.21
Relay pages all Ok still no issues at all.
Regards
Rob D
NZ

kwindrem, 2021-09-29:
Rob, thanks for reporting this. The issues have nothing to do with the MK3 USB interface. I'm using files for my modified enhanced flow (and Tile Overview) when I should not be. The missing information is due to the way the overview code passes parameters to the child objects. This will be corrected in the v3.5 release.

RobDuthie, 2021-10-09:
Just an update on inverters with VE bus  inverters inteface connected, It still clears your power bars and enhanced flow  overviews when connected.
Do you need any more information data, screen shots etc from me for testing this issue?
It's running your latest versions and  Venus is 2.80.21.
Regards
Rob D
NZ