kwindrem/GuiMods #19: Show multiple strings MPPT in dashboard

nhybrow, 2021-10-04:
Hi,
I have three strings on three MPPT in my system. Is it possible to show the three strings in the dashboard ?
Other thing, in the battery panel, the gauge of remain capacity protrudes on the left about 1 or 2 pixels.
Thanks for your job!
Ben from France

kwindrem, 2021-10-04:
There is room for 3 PV Chargers if you don't display temps and tanks. With temps and tanks, there is only room for 2.

I've seen the bar for actual current extend past the ends of the gauge but not the entire gauge. The battery gauge extending past the tile usually means the limits are not set correctly. That is, the actual battery current is exceeding the charge and discharge limits. Refer to the ReadMe for configuration details.

nhybrow, 2021-10-05:
Hi,
Regarding the three solar regulators, I cannot see all three, although the temperature and tank sensors are disabled. I still only have one regulator in the yellow part.
Otherwise, concerning the current gauge, I do not see or adjust the limits so that it does not protrude on the right. The information concerning my battery, and the operating limits are in my opinion correctly fulfilled. Can you guide me on this point?
thanks in advance

nhybrow, 2021-10-05:
Another thing, which temperature sensors do you use for which are displayed in venus OS?

nhybrow, 2021-10-05:
You can see the gauge problem here
[https://photos.app.goo.gl/qLevbysaiEYwWpmHA](url)

kwindrem, 2021-10-05:
The screen shot of your battery gauge suggests the charging limit is set to zero. As explained in the ReadMe, this setting is in the DVCC setup:

Device List / Settings / DVCC / Maximum charge current. DVCC and Limit charge current must both be on to see these settings. If you are not using DVCC, then turn it on, make the setting then turn it off again. The charge current limit will persist even with DVCC off.

I don't want to duplicate that setting in the GuiMods menu because it affects system behavior, but am considering adding a separate parameter for the gauge since several people have been confused the way it is now.

The other gauge settings are in Device List / Settings / Display & language / GuiMods / Power Gauges. The ReadMe explains what each is. 

Temp sensors: 
https://www.victronenergy.com/accessories/temperature-sensor-qua-pmp-venus-gx 
or you can make your own with an LM335
Also with v2.80 (still in beta) the RuuviTag and RuviTabPro sensors:
https://ruuvi.com/ruuvitag/

nhybrow, 2021-10-06:
Ok thanks for that aswer. I think its good now.
But my initial prob about mppt chargers keeps going. I can see only one of my three solar chargers, and as you can see, the only one I see is displayed twice... How do you explain that ?

kwindrem, 2021-10-06:
What version of Venus are you running? I checked and prior fo v2.73, I only display two PV chargers. I'm changing that now.

An older version of GuiMods had a bug that displayed the first PV Charger twice but I thought I'd fixed that. Checked the code and it appears to be correct.

GuiMods v3.11 will have that and another change.

kwindrem, 2021-10-06:
I have updated GuiMods to show 3 PV chargers for that venus version now.

If your GX device is connected to the internet, go to Settings / Display & language / GuiMods / Package Versions and updates and select Check packages once or Fast one pass if you want to keep packages updates automatically.

You can also use the online updates for Venus firmware to get to the latest (currently v2.73).

If you r GX device is not connected to the internet, you can follow the instructions in my ReadMe files for downloading an update to my packages.

Venus firmware downloads can be found here:

http://updates.victronenergy.com/feeds/venus/

You want the release feed then images, then choose your platform. "einstein" is for Cerbo. "beagle bone" is for Venus GX and OctoGx. "nanopi" is for the Multiplus-II GX and EasySolar-II GX. raspberrypi2 is for PI 3 also. Others should be obvious.

nhybrow, 2021-10-09:
Hi,
I have finaly reinstall my venusos on my raspberry with v2.73 image. I have reinstall your mods, and now I have three line in PV display. But there is an ather prob.  The voltage value are at 0.01v on the three ones, and one of the three PV regulator has his Amp at 0A while its producing. What can I do to repair this prob.
Thanks in advance ;)

kwindrem, 2021-10-09:
I found some more bugs that may or may not have created these issues. Upgrade to GuiMods v3.13 and see if that fixes the problems. If not, please provide a screen shot.

Are you using Victron PV Chargers? If not, I wonder if your units are providing valid voltage, power and current.

nhybrow, 2021-10-09:
Perfect, it works great !!
Thanks