kwindrem/GuiMods #21: GuiMods ver 2.80.24

RobDuthie, 2021-10-16:
When using the enhanced over flow view, it blanks out the main over view page. When tunred off it appears OK again.

sasfs2, 2021-10-16:
> Fixed in v3.15

Still blank with v3.15

sasfs2, 2021-10-16:
Kevin, still getting blank page with v3.15

kwindrem, 2021-10-16:
It's working here so I'll need some additional information. Type the following at a command prompt of your Venus OS device and reply with the output

tail -40 /var/log/gui/current|tai64nlocal

RobDuthie, 2021-10-16:
Doesn't blank out anymore thats is fixed ok, but if using mK3 interface with inverters like before the overflow does not show up.
VE Direct is all OK.

kwindrem, 2021-10-16:
On GuiMods, not GuiMods2, if it’s an ESS, the GridParallel overview is used and this one does not have my enhancements. Is the inverter at the top of the screen  indicates Hub Overview) or more towards the center (indicates Grid Parallel overview)?
—
Kevin

kwindrem, 2021-10-16:
> It use to work ok on the GuiMods ESS grid parallel system with enhanments Inverter is in the center of the screen.
> I have installed the GuiMods2 as test this works OK with enhancements.
> Confused now.

If the overview is showing the inverter more in the center of the screen, this is the grid parallel overview and I have never applied my enhancements to that version. Because which overview is shown depends on the system type, it may be that conditions were such that it switched to the Hub Overview screen. This would most likely be the case if the system is not configured for ESS.
—
Kevin