kwindrem/GuiMods #28: Translation of the setting descriptions

lennycb, 2021-12-16:
Will there be a translation in a later version? Should mean that all "setting descriptions" adapt to the set system language.

kwindrem, 2021-12-16:
Yes, I'd like that also but not sure how it is accomplished. There are hooks in the QML code (strings are wrapped in a qsTr () call, but there is an off-line process that creates the translation database that I don't know anything about. If you know, please pass on that information.

lennycb, 2021-12-16:
You are right. Should be this file/s. --> /opt/victronenergy/gui/lang

The file probably needs to be compiled. With the last update the associated .ts files were also deleted. (Presumably the compiling base)

A (very) small excerpt from the old file color_control_de.ts:
`<message>
        <location filename="../qml/PageSettingsDisplay.qml" line="65"/>
        <source>Show tanks overview</source>
        <translation>TankÃ¼bersicht anzeigen</translation>
    </message>
`
Unfortunately, I'm out when compiling. No nerve for that.

lennycb, 2021-12-17:
Update:
you can change it. But it's really tough work.
Here is an example. Spanish and German.
I just added "- Deutsche Übersetzung
[image]
[image]
" to the German setting for "GUI-Mods"

kwindrem, 2021-12-17:
I'm willing to do the work but don't know where to start.

Are there instructions anywhere?

Is this something I can get to or is it Victron internal?

lennycb, 2021-12-17:
So I don't have any instructions, but I can tell you how I did it.
PS: your language? Maybe then I don't always need to hunt it down in the Google translator :-)

The .ts file is no longer included in the current version. So you have to take this from the last version.

As an example using your file PageSettingsGuiMods.qml

1. Searched for everything in the file that is within qsTr ("xxxxx")
2. made a translation for this
3. The gui_de.ts file is expanded as follows
```
<context>
    <name>PageSettingsGuiMods</name>
    <message>
        <source>Gui Mods</source>
        <translation>Einstellungen GuiMods</translation>
    </message>
	.
	.
	.
	.
	.
    <message>
        <source>Inactive Tiles on Flow Overview</source>
        <translation>Inaktive Kacheln in der Flow-Übersicht</translation>
    </message>
    <message>
        <source>Power Gauges</source>
        <translation type="unfinished">Einstellung Leistungsmesser</translation>
    </message>
</context>
```

4. This adjusted file is then opened with the program "Qt Linguist" and a release (gui_de.qm) is created.
5. Upload with a file manager (replace file): /opt/victronenergy/gui/lang/

You then have to do this for all languages ​​and above all for all of the .qml files you have created.

And now comes the crux of the matter. You would have to change or exchange the translation files when installing. But since you are working with a .ts file that will become obsolete over time, there will be problems if Victron changes something in the .ts.

It would be better to register your changes as "pull requests" at Victron.

kwindrem, 2021-12-17:
Thank you for the detailed explanation. I do have the ability to override the .qm files on a version by version basis, BUT can’t do it separately for each package, but maybe just for GuiMods.

As you say it’s LOTS of work. Plus I’m not familiar with any other language so I’d need someone else to check my work

I’ll see what Victron says about incorporating GuiMods in their translation. Otherwise, probably not going to happen.
—
Kevin

lennycb, 2021-12-18:
I have already finished 3 files:
PageSettingsGuiMods.qml
PageSettingsGuiModsGauges.qml
DetailBattery.qml

if you want to try the German translation, copy the file into the folder and reset it.
gui_de.qm copy to /opt/victronenergy/gui/lang/

The file name was changed when the version was changed. Then just adjust the name.

[gui_de.zip](https://github.com/kwindrem/GuiMods/files/7739378/gui_de.zip)