kwindrem/GuiMods #32: Little thing when updating from 5.5 to 5.7

lennycb, 2022-01-16:
Hello, I just updated from 5.5 to 5.7. I noticed one small thing.
At the end of the installation you would otherwise be asked for "Restart GUI" --> NOW or LATER.
Now it said "Installing GUI" --> NOW or LATER

Is this just a display error on my part?

Jsrivet, 2022-01-16:
This is the what language transcription I am in French and GuiMods always in English

kwindrem, 2022-01-16:
I'll look into the Restart GUI vs Installing GUI issue. It should not be like that

As far as language, there are no translations for GuiMods or any of my packages. Sorry.

lennycb, 2022-01-16:
@ Jsrivet So this will continue to be in English until someone does a translation of it. In the gui_fr.qm (or in other translation files) the part of GuiMods is not included. The cooperation of "native speakers" is certainly required here. Patience...

kwindrem, 2022-01-16:
The "Installing GUI" message when "restart GUI needed" message is cosmetic only and will be fixed with the next release.

Thanks for bringing it to my attention.

Jsrivet, 2022-01-16:
@kwindrem If you want I can translate into french. I am a french from Canada Quebec

kwindrem, 2022-01-16:
@Jsrivet, I welcome that effort. Thank you. I'm still trying to figure out what happens once someone builds the translation files. It appears they need to be merged into the main language files and not sure how that could be managed without Victron's help. So if you do the translations now, it may be some time before they can be implemented. You might want to hold off until I figure out if your efforts would make it into the system.

lennycb, 2022-01-17:
@kwindrem yes, the new and existing translation must be merged. But if an update comes, everything is gone.
One possibility until permanent implementation by Victron would be to change the translation file manually after each update (or by script)
Has advantages and disadvantages:
Advantage:
- you have great extensions in your native language

Disadvantage:
- If Victron changes something in its programming/translation, you change the correct translation into an incorrect one (the change is not yet included in the manual translation file)
- If you change something, kwindrem, the new/adjusted part will remain in English until the translation file is changed.

In order to get the thing implemented at Victron, nothing serious should change at GuiMods.

@kwindrem are you actually finished or do you have any other great ideas for GuiMods?

PS: German is 85% finished :-)

kwindrem, 2022-01-17:
Thanks Lenny. I really appreciate the time you are putting into all this.
I have updated GuiMods with your latest changes.
Unfortunately, GuiMods is an evolving thing. There are still things on my ToDo list for it. I may be adding tiles to the flow grid parallel overview page for alternator and/or DC-DC converters, and will most likely add a DC System details page. Then who knows what will be next.

You say "must be merged". Is that something I can do, then include the merged version in my package(s)? I can create modified files for each Venus version so doing the same for translation files wouldn't be a problem.

lennycb, 2022-01-17:
So basically it's more of a "fumbling job".

there is a base file. let's say it's called gui_en.ts. In it are all qsTr("....") from all qml files. Unfortunately we don't have that. So there are 2 possibilities:
1. we build one as a sample
2. each individual gui_xx.qm must be converted. .sqm --> .ts
(backward conversion is always a bit stupid, sometimes something goes wrong)

Suppose we have built a pattern and there is a new .qml file with lots of qsTr("bla blah") in it,
then this .qml is added to the pattern
--> lupdate qml/new file.qml -ts base/gui_en.ts

Then I mostly translated manually with the Linguist program (I use 6.2.2). Since we do not yet have the PhraseBook for the new translations, it will be created with each new translation.
Advantage, if we have a word in the PhraseBook, a later translation can be done in batch.

Finally, release it as gui_de.qm or in the translated language and exchange it with the original.
--> /opt/victronenergy/gui/lang

So what can happen in the worst case: Victron changes something in the qml files and you don't notice. The user does a system update, the SetupHelper reinstalls all mods and overwrites the translation file (if you want to automate). Then the new change from victron remains in English because the system does not find an entry. And whoops you have another issue. :-)

I don't know if that's the "normal" way to work. But that worked fine for me. And up until 2 weeks ago I had nothing to do with it. :-)

kwindrem, 2022-01-17:
Where do I get Linguist and any other tools needed to work on the translation files?
I use Mac OS but if it's Windows only, I do have that under Parallels.

You say it is possible to backward convert the .qm file supplied with Venus OS but didn't say which tool.

lennycb, 2022-01-18:
@kwindrem I sent you a link to my repo.