kwindrem/GuiMods #37: Ruuvi temperature "out of range" alarm ?

mjelkins, 2022-01-30:
Not really an issue but...
1# Would be nice if you could see the additional info when touching a (Ruuvi) temperature box - ie then show Humidity - etc - similar to touching the MPPT or AC Loads boxes - etc.
2# Could one set up temperature "out of range" alarms? I have four Ruuvi's in my Caravan, Inside (the caravan), Outside, Freezer, Fridge. Others may have sensors near batteries (though I'm happy with the wired Victron temperature probe). For example, if the Freezer goes above "-2°C" - then switch on the screen and flash the Display. Maybe make an alarm sound(?). Also allow a relay to be triggered in an alarm event - e.g. to switch on the electrical "warming" pad wrapped around a Lithium battery when a battery gets down to 3°C. Also needs an "Off" switch for when Fridge/Freezer is switched off.

ps - I absolutely love the software that you have provided!!

kwindrem, 2022-01-30:
Touching the temperature tiles is reserved for scrolling when the number of temps exceeds what can be displayed, so no option to touch for more details. Would be nice but can't do.

Venus v2.80 (currently in beta) provides temperature based alarms. I think it's just closing a relay when temps are out of bounds. I don't think there is anything on screen. I think I can use the temp alarm settings to make a change to the temp tile.

mjelkins, 2022-01-30:
OK - no problem with tile touching. I bought the Ruuvi to see the temperature anyway

Simply changing the colour (or similar) would be a good addition!

rgronachon, 2022-10-23:
Hello,
I am new to the site, but am utterly impressed with the work you have done!. I am wondering what it would take to add a toggle that would allow us in the United States to read the Ruuvi’s in Fahrenheit or Celsius?

kwindrem, 2022-10-23:
Venus OS v2.90 added a system-wide temperature preference. It's in the Display & Language / Units menu.

GuiMids overview pages, as well as the stock device list menu items use this preference. Prior to the system-wide temperature preference, GuiMods provided one for the flow overviews but this one did not affect the menus.

The only item I know that does not use the system-wide temperature preference is the Mopeka tank sensor menu. I'll leave this to Victron to fix in a future firmware upgrade.

closed 2022-10-23 by kwindrem