kwindrem/GuiMods #38: GuiMods file set incomplete

JensDue9, 2022-02-01:
Hallo
When I try to install from within Package manager it comes with this message: GuiMods file set incomplete.
( RaspberryPi 3+ running v2.80~21-large-23 )

Jsrivet, 2022-02-02:
I have the same problem with my Cerbo GX v2.80-41-large-25

kwindrem, 2022-02-02:
Apologies. There was a bug in SetupHelper. Fixed in v4.6

The issue was that I wasn't checking the file set for a released version for matching files. Since v2.8...large is still in beta, no matching file was found since the match was in v2.80.

If you don't have automatic download and update on, download SetupHelper v4.6 and install it. All the packages should then install.

virtuvas, 2022-02-05:
none else commenting,

still see incomplete file set when i try to upgrade GuiMods.
I'm on large and on 5.10 GuiMods and 4.7 SetupHelper.
let me know if you want any logs.
V.

kwindrem, 2022-02-05:
GuiMods v5.10 does not support v2.80, including the large version.
You need to upgrade it to the latest which is v5.13

virtuvas, 2022-02-05:
I'm on v2.80~33-Large24. 
and I cannot upgrade from 5.10, that's what I'm commenting. Not sure why, ideas?

kwindrem, 2022-02-05:
What does PackageManager show for the three GuiMods versions?

Going into GuiMods in the active list, what are the GitHub user and branch/tag settings?

How are you attempting an update? automatic, manual from the PackageManager menu, manually from the command line?

virtuvas, 2022-02-05:
[image]

.

[image]

.
[image]

tried a couple of times to download it again, no errors on downloading, just same error when trying to upgrade. I've disabled the autoinstall

sasfs2, 2022-02-05:
Same issue after downgrading from 2.81 to 2.80

kwindrem, 2022-02-05:
You should run this command:

`rm -rf /data/SetupHelper-blind
`
then uninstall and remove the SetupHelper-blind package shown in PackageManager. It's possible you actually have two different versions of PackageManager running which would could cause all sorts of issues.

I have tried installing GuiMods v5.13 on Venus v2.80, v2.18 and v2.80-large-26. All succeed with no errors.

The error usually indicates one of the files I modify has already been modified. The easiest remedy is to install the Venus software and try the GuiMods update again.

If I can get logs I can identify which file is causing the problem or possibly a different issue.

virtuvas, 2022-02-05:
interesting. No SetupHelper-blind in /data.
tried a few times to uninstall it, no luck. now slightly stuck on this:

[image]

not keen on reinstalling Venus now on this rpi.
which logs would you like?

kwindrem, 2022-02-05:
tail -50 /data/log/SetupHelper | tai64nlocal
tail -50 /data/log/PackageManager/current | tai64nlocal

to remove the active packages database:

svc -d /service/PackageManager
/data/SetupHelper/removeManagerSettings
svc-u /service/PackageManager

reboot

PackageManager will rebuild the database from scratch based on packages it finds in /data.

virtuvas, 2022-02-05:
[log1.txt](https://github.com/kwindrem/GuiMods/files/8009033/log1.txt)
[log2.txt](https://github.com/kwindrem/GuiMods/files/8009034/log2.txt)

stopped the service, run the remove manager settings and restarted service, rebooted. All fine.
Now SetupHelper-blind is not listed in package manager, which I guess is good.
Still installing the GuiMods doesn't work, same error message

V.

kwindrem, 2022-02-06:
The Venus OS version you are running is an older beta version and there is no file set for it in GuiMods.

You should update your system to the latest Venus large which is v2.80-large-26.

virtuvas, 2022-02-06:
thanks Kevin,

maybe an edit to the "GuiMods File set incomplete" to give more feedback (and avoid more questions like mine!)?

cheers
V.