kwindrem/GuiMods #39: Suggestion: PV Display options

Albarge, 2022-02-01:
Hi @kwindrem , thanks for GuiMods!

I wonder if it's possible to give the option to display the PV box with abbreviated MPPT name and power (watts) on the same line. Currently it has the name above and V/I below, I can see the benefit of both for different situations though.

Personally I have 2 x Ve.can Mppt's and 1 x Ve.direct, but only the ve.direct Mppt name displays anyway, so having a line for the name isn't useful for Ve.can devices currently, also with the Ruuvi's on the bottom row, the PV box can only show two Mppt's anyway, which uses four rows, so in theory it could show four Mppt's if we could have name and power on the same line?

I don't know if that's possible, but it would be great to have the option maybe in the GuiMods display settings!?

I think you'll know what I mean, but here's a pic of my display with the cramped PV box:
[image]

Thanks Al

kwindrem, 2022-02-01:
You can currently get detailed information of all PV chargers by taping the center of the PV charger tile.

Albarge, 2022-02-01:
Yes, that's a nice feature, but I would like to see all three Mppt's or possibly four without needing to tap the tile. I would prefer a single line with name and power for each if it meant they'd all display without tapping. Day to day power is more useful for me to view quickly and see what they're doing, but Voltage and Current are good for checking issues. - I understand if it's not something you would spend time on though, but it would make me happy if possible :)

kwindrem, 2022-02-02:
Done

closed 2022-02-02 by kwindrem

Albarge, 2022-02-03:
Oh wow great! Looks like it will be much easier to see the power output of a few Mppt's at once, I quite like the scrolling of the long named ve.direct one too. 
Thanks so much. Do you have a patreon type thing? Or if you're in London I'll get the beer. 🍺🙂