kwindrem/GuiMods #42: Temperatures jumping around randomly with Ruuvi Sensors

triptopolous, 2022-02-24:
Well, to preface this, I'm so thrilled to have GuiMod on my Victron stuff!  Victron has screwed the pooch on Celsius vs Fahrenheit but GuiMod fixed all my problems in just a half hour install.  I'm THRILLED with GuiMod.  However (you knew there was a however), I did notice that the temperature probes were all backwards with the Cerbo GX and GuiMod.  Now worries, I rewired it all and it's fixed.  Then I added a couple of Ruuvi Sensors and they randomly pop up in GuiMod however they are read I guess.  It's making me nuts.  I'm willing to just send my sensors to you and you can have them if you can fix the issue.  Or I'll get you new ones.  You pick.  I REALLY appreciate open source work and benefit from it all the time.  However, the last programming I did was Fortran 77 and that's a LONG time ago.  So, I'm happy to help (financially or with hardware) but I can't code anything! This sounds like a Ruuvi thing for sure as a few folks on the FB forum are seeing it to.  Thank you for your consideration and let me know how I can help.

triptopolous, 2022-02-24:
[image]

triptopolous, 2022-02-24:
[image]

triptopolous, 2022-02-24:
Obviously, it would be nice to have the Fridge/Freezer next to one another.  I like having "Battery" under the battery.  Is there a way we can control the order?

kwindrem, 2022-02-24:
Currently, I build the list of temps in the order they are found in dbus services. This can be different with each boot due to the order devices are discovered.

There is a way to sort the list of temps but I do not know how it is done. Victron did this for tanks a while back.

triptopolous, 2022-02-24:
Thank you for the note.  I'm sure that Victron will end up ordering them somehow as this 2.8x gets rolled out more.  Thanks again for all your work!

closed 2022-02-24 by triptopolous