kwindrem/GuiMods #43: Show humidity in temp tiles

The2Earthlings, 2022-02-24:
It would  be great to display the humidity just after the temperature  if space allows it and ruuvi is able to.

Example: 13.2°C • 53%

kittyorianda, 2022-05-22:
Would be happy if the integration of the humidity indicator from Ruuvi sensors would be possible.

kwindrem, 2022-05-22:
There simply is not enough space for humidity, at least on the "compact" views where name and values need to fit on one line.

kittyorianda, 2022-05-22:
With long names the system scrolls from right to left. Doesn't that work there too?

kwindrem, 2022-05-22:
Yes, but if humidity were displayed there's be about 1 or 2 character spaces for the name to scroll in

The2Earthlings, 2022-05-22:
Hi @kwindrem - in the view there should be enough space: first line "name of the sensor", second line "temperature and humidity, if not enough space in small views you can scroll or hide humidity.

Here an example how it can work

[image]
k

kwindrem, 2022-05-22:
Yes. It's the "compact" version where this won't work:
[image]

I can probably also make humidity fit in the Mobile overview even in the compact layout there:

[image]

But names will need to be fairly short or scroll

kittyorianda, 2022-05-22:
[image]

Thanks @ mine would also have space, even if I plan to add an additional 3 sensors.

Yes, if you have these views, then things get tight. But maybe you can make it selectable in the options, whether you want to have the simple version, or the detailed?

kittyorianda, 2022-05-22:
A clickables field similar to touching the MPPT or AC loads boxes - etc., I read somewhere would probably not go?

kwindrem, 2022-05-22:
The selection between the two-line and single line format is done automatically based on how many tanks and temps need to be shown. I can't make this a manual selection. Your only choice would be to hide tanks, then the temps would be two line in the flow view. In the mobile overview, it would still switch to single line mode if there were lots of temps.

Touch on the tanks and temps tile won't work because these are set up in a scrolling list so if there are lots of them, you can see them all by scrolling. Can't do both tap and drag to scroll.

kittyorianda, 2022-05-22:
On the mobile view an additional third line and make the rest of the display above a little smaller?