kwindrem/GuiMods #48: Analog input not showing up IO

RobDuthie, 2022-03-07:
Re version greater than >2.41 analog tanks don't show up
Re the mssing IIO folder in the sys/bus, this is why the analog ports don't show up at all on the IO page.
I have tried copying across the files from an older ver <28.41 to no avail.?

kwindrem, 2022-03-08:
If the analog inputs are not showing in /sys, this is not something to do with GuiMods. 

My other package RpiGpioSetup doesn't do anything with analog inputs so this issue doesn't belong in that repo either.

RobDuthie, 2022-03-08:
Hi Kevin
I relize that , it is not complied in the main image files from the Venus feeds site, for some reason it is left out?
Why do they do this as it was OK in the older Ver 2.41?