kwindrem/GuiMods #49: blank enchanced flow screen

jengstro2, 2022-03-27:
Hi, using 5.19. If Enhanced screen is turned on, i will get blank screen + the webpage/canvas gets frozen for some time. Until able to get back to menu.

I have "normal" device set charger, shunt + inverter

kwindrem, 2022-03-27:
I need much more information to be able to help you. Are you able to look at logs via a command line interface?

jengstro2, 2022-03-28:
Newbie for Victron stuff... Is it logging all the time? can I see the logs somewhere from rasp/shell (or is this what you meant by command line iface)

I have to say, nice packaging/distribution system... well done 👍 . Bit strange in the beginning, I installed the SetupHelper from ssh/shell command line. Not via ssd card tricks.

jengstro2, 2022-03-28:
Here is at least installation log
```
2022-03-25 21:13:17.423677500 GuiMods/setup: creating package options directory /data/setupOptions/GuiMods
2022-03-25 21:13:19.010886500 GuiMods/setup: creating GuiMods Settings
2022-03-25 21:13:19.814221500 GuiMods/setup: installing GuiMods files
2022-03-25 21:13:20.669761500 GuiMods/setup: ++ GuiMods installed
2022-03-25 21:13:20.696277500 GuiMods/setup: adding GuiMods/setup to reinstallScriptsList
2022-03-25 21:13:20.721273500 GuiMods/setup: GUI restart needed
2022-03-25 21:14:36.525213500 SetupHelper/reinstallMods: reinstallMods starting
2022-03-25 21:14:37.532361500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-25 21:14:40.355437500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-25 21:14:44.301418500 SetupHelper/reinstallMods: reinstallMods complete
2022-03-25 21:21:31.529353500 GuiMods/setup: ++ GuiMods uninstalled
2022-03-25 21:21:31.548636500 GuiMods/setup: removing GuiMods/setup from reinstallScriptsList
2022-03-25 21:21:31.575758500 GuiMods/setup: GUI restart needed
2022-03-25 21:14:36.531117500 SetupHelper/reinstallMods: reinstallMods starting
2022-03-25 21:14:37.597838500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-25 21:14:40.329358500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-25 21:14:44.040411500 SetupHelper/reinstallMods: reinstallMods complete
2022-03-26 07:42:36.539638500 SetupHelper/reinstallMods: reinstallMods starting
2022-03-26 07:42:37.538039500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-26 07:42:40.360684500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-26 07:42:44.026289500 SetupHelper/reinstallMods: reinstallMods complete
2022-03-26 07:42:36.523698500 SetupHelper/reinstallMods: reinstallMods starting
2022-03-26 07:42:37.565703500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-26 07:42:40.366508500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-26 07:42:44.065794500 SetupHelper/reinstallMods: reinstallMods complete
2022-03-26 07:42:36.522834500 SetupHelper/reinstallMods: reinstallMods starting
2022-03-26 07:42:37.560032500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-26 07:42:40.337685500 SetupHelper/reinstallMods: waiting for dBus settings
2022-03-26 07:42:43.993378500 SetupHelper/reinstallMods: reinstallMods complete
2022-03-27 12:10:24.692623500 GuiMods/setup: installing GuiMods files
2022-03-27 12:10:25.756635500 GuiMods/setup: ++ GuiMods installed
2022-03-27 12:10:25.784207500 GuiMods/setup: adding GuiMods/setup to reinstallScriptsList
2022-03-27 12:10:25.810130500 GuiMods/setup: GUI restart needed

```

jengstro2, 2022-03-28:
Bit looked more deeper that "web page". Weird Canvas / QT? framework to draw couple of boxes and some animations. I just wonder, why not HTML(5) and css... (js) so much scaleable etc.

kwindrem, 2022-03-28:
The log I need is for the gui:

tail -50 /data/log/gui/current | tai64nlocal

I don't have any insight on why the GUI was written in QML. That was a decision made by Victron long, long ago when HTML5 didn't exist. The main GUI is not done with a web browser, rather it is displayed on the local screen.

jengstro2, 2022-03-29:
Got it: 
2022-03-29 06:13:00.167441500 INFO  VeQuickView::keyPressEvent: wakeupEvent
2022-03-29 06:13:58.617002500 file:///opt/victronenergy/gui/qml/OverviewHubEnhanced.qml:953:16: TankModel is not a type

I do not have any Tank configruations (my setup is garage, not marine )

kwindrem, 2022-03-29:
This is odd because the error indicates a stock file from Venus OS is not being detected or is corrupted: /opt/victronenergy/gui/qml/TankModel.qml

What version of Venus OS firmware are you running?

I am not seeing the error on my systems which are running 2.83

jengstro2, 2022-03-30:
Spot on!. The TankModel.qml is missing. Weird... anyway I'm running 2.66 (latest release, I presume). I can see your gui mod (add-on) files there also.

I have not delete the file separately myself.... is the SetupHelper modifying the existing QT files or just adding new ones? 

What is left to here to try... is full SD card reinstall ( argh :-) ). Bit inspect what happens, and can we repeat this.
Anyway, before doing full reinstall.. is there anything we can dig in still?

jengstro2, 2022-04-12:
Back to finalize this issue. From (2.66) to upgrade over firmware over >=2.7 couldn't be done without doing full SD reinstall.
Anyway, running now on 2.81, GUI mods are working nicely. This package manager is awesome, without this I would not know that there is even BMS123 support for venus-os.  Small thing for the waveshare 7inch hdmi touch display, it requires some more config.txt settings... I can put this issue on RpiDisplay side, in case there is no story about it.  Anyway, everything is working nicely now.