kwindrem/GuiMods #50: Two digit decimal for battery voltage in Flow panel

GeoZBee, 2022-03-31:
I have been using guimods for a little while now and absolutely love the features it adds to my CCGX console.
One thing I would like to see is 2-digit decimal precision on the battery voltage display, in the "Flow" panel on the CCGX. While one digit precision is likely sufficient for lead-acid batteries, with many lithium batteries, a few hundredth of a volt can be significant.
I don't know why Victron doesn't just do this by default. The VRM portal console is already two-digit decimal for nearly all voltage readings.

Thank you.
 GM

kwindrem, 2022-03-31:
Only under very controlled conditions is battery voltage to that precision of any use. For lead-acid batteries, the battery must have been at rest for a period of time without any charge or discharge during the resting period. For lithiums, load on the battery will drop its terminal voltage so the only time any level of precision is of value is when the current in/out of the battery is near 0.

GeoZBee, 2022-04-08:
Hmm,
My batteries are accurate to the 100th of a decimal, and I can usually tell by voltage and load how much charge capacity is in the battery, and how stable it is. Granted, it rips though that second decimal point pretty fast as the capacity drops with load. 
And those numbers are provided, at that level of detail by the BMV and the online VRM portal, but not on the GX for some reason.
So, I would still find it useful, but... we'll have to agree to disagree :)

I would modify the QML file directly, but suspect it would likely get clobbered by a future update.

seamaster101, 2022-10-05:
I agree. 2 decimal display is a "must"as lithium batteries are monitored in mV -3 decimals

virtuvas, 2022-10-07:
sorry, slightly confused, I do get 2 decimal points in 2.90. Granted not from BMV but the BMS but it's all shown on the normal "flow" gui dash.
So, looks like it's fixed, should be closed as a issue?

V.

GeoZBee, 2022-10-18:
This issue seems to have been fixed in current updates.

closed 2022-10-18 by GeoZBee