kwindrem/GuiMods #51: Getting several CCGX reboots while navigating --post Guimods

GeoZBee, 2022-03-31:
After I installed guimods, I'm noticing that my CCGX is occasionally rebooting on its own. This usually happens whenever I try to switch back and forth from the Flow panel to the settings page. Sometimes hitting the center button (several times) while in the Flow panel will trigger a reboot. And I can repeat it with some regularity, if I try to use the "details" options on the individual display elements by tapping on them. While "tapping" obviously doesn't exist on a CCGX, I'm able to simulate a tap using the remote console from VRM.

This is far from a high priority issue --as left alone the unit is pretty stable, but is more of an FYI.

kwindrem, 2022-03-31:
The additional data presented on the enhanced flow page does increase the CPU load and switching back from the menu to the flow overview is very slow. It is possible that a CPU overload could be triggering a watchdog timer reboot but I have yet to see this and this is the first complaint.

It also might be possible that your CCGX has a mechanical intermittent and pressing buttons could flex the circuit board enough to cause the reboot. I think this is more likely than a software-related problem because the center button only brings up the bottom strip with the menu and pages buttons, so very little additional CPU load.

philipa, 2022-03-31:
FWIW my Cerbo has started to reboot at random. Perhaps every 8 hours or so, but I've not discerned any pattern yet.

I'm going to run for a few days without guimods.

kwindrem, 2022-04-01:
If you have terminal access, take a look at CPU loading.

The command uptime or top will show you load averages for the last 1 minute, 5 minutes and 15 minutes, respectively. A value of 1.0 indicates 100% CPU usage for a SINGLE CORE. Cerbo has a dual core CPU, CCGX has a single core CPU.

top also shows memory utilization which might also be a factor in running a heavily loaded CCGX.

The enhanced overviews add a significant amount of additional information and hence impact the CPU loading. But I have not seen huge increases in CPU load averages. Each of the enhancements can be disabled:
Enhanced Flow overview, Relay overview, Enhanced Mobile overview, gauges. Two different flow overviews are provided. The Enhanced Grid Parallel overview actually uses less CPU power since it does not include all the details provided by the other version.

GeoZBee, 2022-04-08:
I'm fairly certain my reboots are related to Guimods. The unit was stable before, and I was constantly using the display.
But post-guimods clicking into and out of settings one or two times is often enough to get it to reboot. Length of time between those events seems unrelated. I can be in settings 2 seconds or 20 minutes, when I click to go back to the flow view, it will reboot --usually by the second time.

As was previously said, however, the enhancements added by guimods win over the reboots --for now. But it can be problematic in an ESS system when the CCGX reboots, and the whole system flips-out while waiting for the CCGX to reboot.
Does disabling the other overviews (not in use) reduce the load? I'm currently only using the Enhanced flow overview and have no use for the others.
I'm traveling at the moment, and cannot access the unit. But when I can, I will shell into it and see what top and uptime show.

Honusnap, 2024-04-15:
> The enhanced overviews add a significant amount of additional information and hence impact the CPU loading. But I have not seen huge increases in CPU load averages. Each of the enhancements can be [...]

Hi,
Maybe it has already been discussed, can you decide at which frequency data are fetched and displayed ? If yes, could ba e good idea to have an option to decide frequencies.
I do not know where the time is wasted to calculate datas / display them. I worked for a web agency and we created our own framework, we used a caching system that would use a timer for each component of the web page, let's the clock is updated every minute so when we call the clock module, it check the age of the cached html send it back if still under a minute and call a new built of this cache if not. It was very efficient, part of the webpage that needed update like once a day did not need frequent calculus.
Wouldn't it be efficient (depend if the system file is fast enough) to cache some part of the display... let's say .. i would like the battery Data to be updated once every 30s, the PV box ... every 5s, the AC load ... 5s ... etc .. 
Maybe it could drastically reduce Load ..

kwindrem, 2024-04-15:
I have been running GuiMods on a Cerbo for many years and have never had a random reboot.

Specifying update rates is not possible. I believe the QML code is largely change based so those parameters that are not changing will not result in QML code executing anything. Systemcalc runs every second and provides summary information for most tiles on the flow page. It also seems like other dbus updates are at about that same rate.

The "GuiMods simple" flow overview is the most complex with it's "show every number in the world" concept. You could experiment with the AC or DC coupled flow pages as they show less data, and/or turn off the tank and temp information at the bottom at least to see if that makes any difference.

Honusnap, 2024-04-21:
> I have been running GuiMods on a Cerbo for many years and have never had a random reboot.
>
> Specifying update rates is not possible. I believe the QML code is largely change based so those parameters [...]

Thanks for that precise answer, you do not control the system rate, you modified the display side of things. As you said, DC coupled go down to 8% CPU vs 15% with GuiMods Simple. Temp and tanks are already off..