kwindrem/GuiMods #53: MIssing detail from SmartSolar RS 450/100

morphias2004, 2022-04-20:


kwindrem, 2022-04-20:
Moved from SetupHelper to here since it's a GuiMods issue not SetupHelper.

RS MPPTs have up to 4 trackers and hence up to 4 sets of voltage and PV power parameters. There is a common (total) PV power parameter as well as an output (DC bus) power parameter. PV currents are calculated from PV power and PV voltage.

Currently the flow overview pages display name, PV power, voltage and current for between 2 and 3 PV chargers on three separate lines (name, PV power, PV voltage/current) and up to 7 with only name and PV power on a single line.

Also, the PV Charger detail page needs separate lines for each tracker.

Working on a design now. Stay tuned.

morphias2004, 2022-04-20:
Legend!

I'm guessing the field that is coming ??V NaNa is because the field mapping is slightly different for the RS?

kwindrem, 2022-04-21:
Yes, for single tracker units it's /Pv/V and /Yield/Power. For multiple trackers it's /Pv/n/V and /Pv/n/P where n is a 0-based number indicating the specific tracker.

V?? indicates my code finds an invalid dbus parameter for voltage.

NaN means "not a number" which results in attempting to use an invalid number in some math (I = P / V in this case). I should be testing to make sure both P and V are valid but I do not.

kwindrem, 2022-04-22:
I have pushed a beta version of GuiMods that supports multiple trackers on PV chargers. You can download this from GitHub or switch the GuiMods branch to "beta" in PackageManager.

The tracker details are shown only on the PV Charger Details page.

The voltage and current on the flow overview tile are replaced by "multiple trackers" since there could be up to 4 trackers and no space to display them. Adding specific tracker in formation to the PV Charger tile is possible with a lot more work and some additional run-time processing.

I have no way of testing the changes to please see if it works for you and report back. Thanks.

morphias2004, 2022-04-22:
Thanks mate! 👍 

It will be a few weeks before I have the vehicle back to test it out, but I'll update you as soon as I do.