kwindrem/GuiMods #56: v5.22 Update causes a Gui restart loop

flyrmyr, 2022-05-03:
This morning I woke up to my RPi 4b screen stuck in a Gui restart cycle, where it's on and active for about 10 seconds and then loops back to the Victron "Starting..." screen. 
Between restarts, I noticed Package manager had automatically grabbed the newest v5.22 for me sometime overnight. It was displaying the "Gui restart needed" badge, so I tried manually restarting the Gui via the Package manager.
I also tried restarting the Pi, disconnecting my screen, removing sd card etc. No luck.

Removing the GuiMods package ended the reboot cycle for me, so it seems it was definitely coming from the update. I tried to manually install v5.20 via tag `v5.20` and `v5.2` but clicking "Proceed" does nothing for those tags...

Hoping this can be resolved as I love the work you've done on this package! Happy to help test on my Pi if need be.

kwindrem, 2022-05-04:
Should be fixed now in v5.23. If you have auto downloads and installs enabled, the problem will correct itself.

kwindrem, 2022-05-04:
The problem was incomplete file sets so that it was not possible to create a file set for v2.85. The install script would then attempt an install, fail to update the one file that was reported in the log, then reverse the install. The script then found it necessary to restart the GUI after the uninstall. Because the package was not installed and Auto Install was active, PackageManager attempted another install.

The fix I put in last night addressed the incomplete file set so the install succeeded.

This morning, I also added tests to avoid the constant install loop that resulted in the GUI restarts: SetupHelper install script extension now flags an incomplete file set if a file replace operation fails. Package Manager also now checks for the INCOMPLETE flag file and does not attempt an automatic install.

closed 2022-05-04 by kwindrem