kwindrem/GuiMods #57: PV Charger Detail is empty after installing v5.23

t440142, 2022-05-05:
[image]
[image]
Hi,

Can you have a look at this?

Thanks

Best regards
Peter

Firmware
PV Charger: 1.59
Cerbo GX: 2.84
SetupHelper: 4.13
GuiMods: 5.23

kwindrem, 2022-05-05:
I'm not sure what is going on because the same mechanism I use to discover PV chargers in the overview is also used in the detail.

Have you rebooted the GX device?

Did this issue show up recently or has it always been like this?

Logs would help. In this case it would be the gui log. SetupHelper v4.13 adds a log collection mechanism in the PackageManager backup/restore settings.
Do the backup then remove the media. Plugging the media in a computer you'll see a logs folder. Zip that up and post it here.

t440142, 2022-05-05:
Hi,

V5.20 works as expected, all informations are displayed.
V5.22 cannot be installed, the error message is "GuiMods fileset incomplete".
Then I tried v5.33 again, the detail display is again incomplete as shown in the screenshot.

I will try to establish an ssh connection to access log files. Have to read how to set it up.

Best regards
Peter

kwindrem, 2022-05-05:
I have reproduced the problem and will get a fix posted today. Thanks for reporting.

kwindrem, 2022-05-06:
Should be fixed now in v5.24

The problem was I had included code to handle PV chargers with multiple MPPT trackers. That code has yet to be tested but got accidentally merged into the main branch when I added support for v2.90~3. It's still in place but I have tested it with a single PV charger with a single tracker.

t440142, 2022-05-06:
Everything works as expected again, thanks for your support.

closed 2022-05-06 by t440142