kwindrem/GuiMods #58: PV Voltage and PV Current are missing on OverviewMobileEnhanced.qml page

gharshaw, 2022-05-10:
@kwindrem 
I looked at the code in OverviewMobileEnhanced.qml. It appears that for Current, the code is calling for an invalid dbus item 
"VBusItem { id: pvCurrent; bind: Utils.path(pvChargerPrefix, "/Pv/I") }" At least I don't have that value on my DBUS for my SmartSolar Charger.
I  believe we need to calculate for the current by dividing "/Yield/Power" by "/Pv/V" . Does this sound correct?
[image]

Regards,
Gene

kwindrem, 2022-05-10:
You are correct. Victron dropped the current parameters a while back and I thought I'd made these changes but missed this one. I'll get on it.

kwindrem, 2022-05-10:
fixed in v5.26

closed 2022-05-10 by kwindrem

gharshaw, 2022-05-10:
That was fast.
Thank you