kwindrem/GuiMods #60: GuiMods on Raspi 4 Ver 1.4

RobDuthie, 2022-06-12:
Hi Kevin

Being testing the latest Venus candiate version 2.90.12 on a raspi4 B+ Ver 1.4 PCB and its all works OK. only thing not working or missing is the analog settings page.
PageSettingsio.qml file is there but empty so loaded a copy from a rapsi B+ version to make it show up, this normaly works OK on the raspi B+ version every time up to version 2.87 release versions OK.
But for the raspi4 it does not show up.
So what else triggers this to show up?  Tyring to track down what control what on the IO page.

kwindrem, 2022-06-12:
I seem to remember a bug that hid the I/O page if the first analog input was not defined. I thought that had been fixed and in v2.90 the test is completely different. Is this the problem you are having?

I checked and PageSettingsIo.qml file is not empty on my systems. This is not a file GuiMods or any of my other packages changes.

In the I/O menu, each submenu is shown/hidden based on the number of devices of that type found. In the case of analog inputs, a model is built of all com.victronenergy.adc devices and if that model is empty the submenu won't be shown.

This is not something GuiMods or any of my other packages changes either.

RobDuthie, 2022-06-13:
Hi
Is the raspi4 image compiled different to the raspi3?
When i do a new image file the  PageSettingsIo.qml is always empty on the raspi3 and raspi4 image files, by replacing it with one i  know thats works OK, the analog ports show up on the screen as noramal again but for the raspi4 it won't do this?
Nothing to do with your GuiMods, Victron is leaving something off?
Just tying  to find out how the analog files are all linked so i can compare them.
As least Vicrton has the raspi4 version Ver 1.4 -1.5 PCB working now.
Regards
Rob D
NZ

kwindrem, 2022-06-13:
PageSettingsIo.qml file is not empty on my RPI 4 running v2.90~12. I also looked at a new image created from the Victron feeds.

There are some differences between the RPI 2/3 and RPI4 builds but I don't know what they are.

RobDuthie, 2022-06-13:
Hi

Re missing analog files.
Found this files, folder missing from the raspi4 in /sys/bus/ folder that will explain  the only thing i can't copy it across to the  raspi4 no permissions?
 ls  /sys/bus/iio/devices/iio\:device0

kwindrem, 2022-06-13:
I think everything in /sys is created by the operating system. I don't have any analog I/O on my RPI 4 so not sure how this works. But at some point it's going to involve a driver/DT overlay for your specific card.

RobDuthie, 2022-06-13:
I have all drivers for the overlays already and they work OK, with the the standard MCP3208 chip set as i use on the raspi  stetson HAT interface i have done, as if gives you the 4x tanks and 4x temps and works very well.
So i wonder why my PageSettingsIo.qml is empty and your ones are not? as i download from the same place. Very odd? https://updates.victronenergy.com/feeds/venus/