kwindrem/GuiMods #61: GuiMods file Set incomplet

kittyorianda, 2022-06-14:
Tried to update GuiMods. Then I received the message "GuiMods file set incomplete". 
I uninstalled the package and tried to reinstall it. Same message. Now I have no GuiMods installed. :(

kittyorianda, 2022-06-14:
Can I download and install the previous version on the Cerbo GX?

kwindrem, 2022-06-14:
Please provide more information: Venus OS version, and if possible the SetupHelper log: /data/log/SetupHelper

Previous versions are available on GitHub as tags.

kittyorianda, 2022-06-14:
By entering the tag I could at least install version 5.31 again. But version 5.33 the system refuses to install. A bug in the German version?

kwindrem, 2022-06-14:
What version of the firmware are you running? That will help me locate a possible file set issue.

I would not think the language would have anything to do with the file set incomplete message.

Can you send me the log so I can investigate the specific file?

tail -50 /data/log/SetupHelper | tai64nlocal

If you don't have terminal access, go to PackageManager Backup & Restore and do a backup. That will write the log files to a USB flash drive. You can then put that drive in a computer and zip up the logs directory you find there. Post that here.

kittyorianda, 2022-06-14:
`@40000000624db7e822987fac SetupHelper-blind/blindInstall: waiting for dBus settings
@40000000624db7ea3ac7439c SetupHelper-blind/blindInstall: waiting for dBus settings
@40000000624db7f20212c10c SetupHelper-blind/blindInstall: waiting for dBus settings
@40000000624db7f60f36dbfc SetupHelper-blind/blindInstall: waiting for dBus settings
@40000000624db7fe044f2c1c SetupHelper-blind/blindInstall: blind install SetupHelper
@40000000624db801185b2544 SetupHelper/setup: creating root setup options directory /data/setupOptions
@40000000624db8011f322afc SetupHelper/setup: creating package options directory /data/setupOptions/SetupHelper
@40000000624db80124a19dec SetupHelper/setup: creating file set for v2.84
@40000000624db80523395774 SetupHelper/setup: installing PackageManager service
@40000000624db80531751454 SetupHelper/setup: adding SetupHelper/setup to reinstallScriptsList
@40000000624db80533535dbc SetupHelper/setup: creating /data/rcS.local
@40000000624db80536e6dcec SetupHelper/setup: restarting GUI
@40000000624db84b1750aa84 GuiMods/setup: creating package options directory /data/setupOptions/GuiMods
@40000000624db84b183b7bcc GuiMods/setup: creating file set for v2.84
@40000000624db84e21dab5bc GuiMods/setup: creating GuiMods Settings
@40000000624db850089f9e64 GuiMods/setup: installing GuiMods files
@40000000624db8510a1a151c GuiMods/setup: ++ GuiMods installed
@40000000624db8510c03c864 GuiMods/setup: adding GuiMods/setup to reinstallScriptsList
@40000000624db8510daf5d54 GuiMods/setup: GUI restart needed
@400000006254539c15e13434 SetupHelper/reinstallMods: reinstallMods starting
@400000006254539d1b0aef7c SetupHelper/reinstallMods: waiting for dBus settings
@40000000625453a330eb7dac SetupHelper/reinstallMods: waiting for dBus settings
@40000000625453aa22bc76b4 SetupHelper/reinstallMods: waiting for dBus settings
@40000000625453b53959c91c SetupHelper/reinstallMods: reinstallMods complete
@40000000625f012615e6a2d4 SetupHelper/reinstallMods: reinstallMods starting
@40000000625f01272395bbcc SetupHelper/reinstallMods: waiting for dBus settings
@40000000625f012e2eae3fac SetupHelper/reinstallMods: waiting for dBus settings
@40000000625f01340cf52d44 SetupHelper/reinstallMods: waiting for dBus settings
@40000000625f014320eb1214 SetupHelper/reinstallMods: reinstallMods complete
@40000000625f02a414cc45ac SetupHelper/reinstallMods: reinstallMods starting
@40000000625f02a515fd4b9c SetupHelper/reinstallMods: waiting for dBus settings
@40000000625f02aa213c2bcc SetupHelper/reinstallMods: waiting for dBus settings
@40000000625f02b03146b154 SetupHelper/reinstallMods: waiting for dBus settings
@40000000625f02b7241b5ad4 SetupHelper/setup: creating file set for v2.85
@40000000625f02bd0c3c35b4 SetupHelper/setup: installing PackageManager service
@40000000625f02be0dd4562c SetupHelper/setup: restarting GUI
@40000000625f02c02b80a66c GuiMods/setup: creating file set for v2.85
@40000000625f02d433f48f5c GuiMods/setup: installing GuiMods files
@40000000625f02d602620dd4 GuiMods/setup: ++ GuiMods installed
@40000000625f02d60588e0b4 GuiMods/setup: restarting GUI
@40000000625f02d606afccb4 SetupHelper/reinstallMods: reinstallMods complete
@40000000625f0e50152df57c SetupHelper/reinstallMods: reinstallMods starting
@40000000625f0e5123507d3c SetupHelper/reinstallMods: waiting for dBus settings
@40000000625f0e58062f3aa4 SetupHelper/reinstallMods: waiting for dBus settings
@40000000625f0e5e04ce1464 SetupHelper/reinstallMods: waiting for dBus settings
@40000000625f0e6a0b1234a4 SetupHelper/reinstallMods: reinstallMods complete
@40000000626325741825b1ac SetupHelper/reinstallMods: reinstallMods starting
@400000006263257525f4647c SetupHelper/reinstallMods: waiting for dBus settings
@400000006263257c2642c2fc SetupHelper/reinstallMods: waiting for dBus settings
@40000000626325821c9f628c SetupHelper/reinstallMods: waiting for dBus settings
@4000000062632591327fd17c SetupHelper/reinstallMods: reinstallMods complete
@4000000062641ab615ca1254 SetupHelper/reinstallMods: reinstallMods starting
@4000000062641ab71765c4dc SetupHelper/reinstallMods: waiting for dBus settings
@4000000062641abe07ce488c SetupHelper/reinstallMods: waiting for dBus settings
@4000000062641ac431b384b4 SetupHelper/reinstallMods: waiting for dBus settings
@4000000062641ad32e078be4 SetupHelper/reinstallMods: reinstallMods complete
@40000000626e325825e3b6f4 SetupHelper/setup: creating file set for v2.85
@40000000626e325837a8e4cc SetupHelper/setup: restarting PackageManager service
@40000000626e32590cb1fa24 SetupHelper/setup: completed
@40000000626e32ab1fad3ecc GuiMods/setup: installing GuiMods files
@40000000626e32ac1465ee74 GuiMods/setup: ++ GuiMods installed
@40000000626e32ac172e70cc GuiMods/setup: completed
@40000000626e32f4225cca84 GuiMods/setup: creating file set for v2.85
@40000000626e32f53a58fd24 GuiMods/setup: installing GuiMods files
@40000000626e32f62fb3bd14 GuiMods/setup: ++ GuiMods installed
@40000000626e32f6328eb5fc GuiMods/setup: GUI restart needed
@4000000062742dcb24bee9ec SetupHelper/setup: creating file set for v2.85
@4000000062742dcb37022164 SetupHelper/setup: restarting PackageManager service
@4000000062742dcc075c3954 SetupHelper/setup: GUI restart needed
@4000000062742f332749caec GuiMods/setup: creating file set for v2.85
[... 120 lines cut ...]
@4000000062a8fd6e30199f1c GuiMods/setup: installing GuiMods files
@4000000062a8fd6f33c1cf2c GuiMods/setup: ++ GuiMods installed
@4000000062a8fd6f35918a04 GuiMods/setup: adding GuiMods/setup to reinstallScriptsList
[... cut ...]
@4000000062a9111b1dc066d4 SetupHelper/reinstallMods: waiting for dBus settings
@4000000062a911222cfd150c SetupHelper/reinstallMods: waiting for dBus settings
@4000000062a911340bc9bd04 GuiMods/setup: creating file set for v2.87
@4000000062a9113e0c056a5c GuiMods/setup: ERROR v2.87 OverviewTanksTempsDigInputs.qml no replacement file
@4000000062a9113e10c4bf5c GuiMods/setup: ERROR: incomplete file set for v2.87 - can't continue
@4000000062a9113e121e5034 SetupHelper/reinstallMods: reinstallMods complete
`

kittyorianda, 2022-06-14:
sorry for a big file :(

kwindrem, 2022-06-14:
that's OK I'll look into the issue.

kittyorianda, 2022-06-14:
Mhhh have no Tank Sensors.... therefore even no Tank Inputs ;)

kwindrem, 2022-06-14:
I think I fixed the issue. Not sure why, but one of the files was corrupt in every file set except v2.90~12!!

You should be able to go back to latest and get the new version (v5.35)

Not sure what your issue is with Tank Sensors. Please explain.

kwindrem, 2022-06-15:
I tested v2.87 and GuiMods installs properly.

This probably means the "original" file has been modified or is corrupted.

For /opt/victronenergy/gui/qml/OverviewTanksTempsDigInputs.qml the original is /opt/victronenergy/gui/qml/OverviewMobile.qml

The easiest way to recover is to switch to the backup firmware, then reinstall the current version (don't switch back to the original version in the backups list).

kittyorianda, 2022-06-15:
ahhhh... works fine! Thanks for the great GuiMods and your great support!

aivatco, 2022-07-07:
hey i hope you can help me too, i think i have nearly the same problem. but i don´t know what to do else.

here my log. 2022-07-07 08:22:57.584358500 SetupHelper/packageAutoUpdater: starting up
2022-07-07 08:25:31.828146500 GuiMods/setup: ERROR: incomplete file set for v2.87 - can't continue
2022-07-07 08:25:38.589249500 SetupHelper/packageAutoUpdater: starting up
2022-07-07 08:30:59.184474500 SetupHelper/setup: removing SetupHelper service
2022-07-07 08:30:59.291112500 SetupHelper/setup: removing SetupHelper/setup from reinstallScriptsList
2022-07-07 08:31:00.251796500 SetupHelper/setup: completed
2022-07-07 08:31:03.835967500 SetupHelper/packageAutoUpdater: starting up
2022-07-07 08:31:07.416181500 SetupHelper/setup: SetupHelper installing service
2022-07-07 08:31:07.470644500 SetupHelper/setup: adding SetupHelper/setup to reinstallScriptsList
2022-07-07 08:31:08.442654500 SetupHelper/setup: completed
2022-07-07 08:32:18.721653500 GuiMods/setup: ERROR: incomplete file set for v2.87 - can't continue
2022-07-07 08:32:25.425882500 SetupHelper/packageAutoUpdater: starting up
2022-07-07 08:43:23.756271500 GuiMods/setup: ERROR: incomplete file set for v2.87 - can't continue
2022-07-07 08:43:30.377315500 SetupHelper/packageAutoUpdater: starting up
2022-07-07 08:50:05.475434500 GuiMods/setup: creating file set for v2.90~14
2022-07-07 08:50:05.725300500 GuiMods/setup: ERROR v2.90~14 PageMain.qml no replacement file
2022-07-07 08:50:06.531061500 GuiMods/setup: ERROR v2.90~14 OverviewMobileEnhanced.qml no replacement file
2022-07-07 08:50:07.260182500 GuiMods/setup: ERROR v2.90~14 PageSettingsDisplay.qml no replacement file
2022-07-07 08:50:07.397957500 GuiMods/setup: ERROR: incomplete file set for v2.90~14 - can't continue
2022-07-07 08:50:07.536807500 SetupHelper/setup: SetupHelper installing service
2022-07-07 08:50:07.578681500 SetupHelper/setup: completed
2022-07-07 08:57:47.821536500 GuiMods/setup: ERROR: incomplete file set for v2.90~14 - can't continue
2022-07-07 08:57:53.257555500 SetupHelper/packageAutoUpdater: starting up
2022-07-07 09:08:45.792733500 GuiMods/setup: ERROR: incomplete file set for v2.87 - can't continue
2022-07-07 09:08:52.459104500 SetupHelper/packageAutoUpdater: starting up
2022-07-07 09:12:00.822331500 GuiMods/setup: ERROR: incomplete file set for v2.90~14 - can't continue
2022-07-07 09:12:00.961050500 SetupHelper/setup: SetupHelper installing service
2022-07-07 09:12:01.003152500 SetupHelper/setup: completed
Choose an action from the list above:

thanks for your work. and help

greetings Armin

kwindrem, 2022-07-07:
Are you using the latest version of GuiMods (v5.32)? If not, you need to upgrade as there was an issue in the previous version that prevents install.

The other reason for these errors is if something has modified one of the files GuiMods does. If that's the case, the install script can't match the current file to one of it's file sets. Swap to the other backup firmware then install the current version (don't swap back). This should put fresh versions of all files in place so the installer can find a match.

Zirrad, 2022-07-21:
Having a similar issue:

 ```
root@raspberrypi2:~#  wget -qO - https://github.com/kwindrem/GuiMods/archive/refs/tags/latest.tar.gz | tar -xzf - -C /data
root@raspberrypi2:~# mv /data/GuiMods-latest /data/GuiMods
root@raspberrypi2:~# /data/GuiMods/setup
creating package options directory /data/setupOptions/GuiMods
creating file set for v2.80~26
ERROR v2.80~26 PageMain.qml no replacement file
ERROR: incomplete file set for v2.80~26 - can't continue
```

On a Rpi
v2.80~26
Victron Energy
20211024095400

BTW Thanks for all your efforts.

kwindrem, 2022-07-21:
Venus OS v2.80~26 is as old beta test version and should no longer be used. I suggest updating to v2.87, the latest released version.

Jsrivet, 2022-09-08:
The same problem occurs with version 2.90-28 I send you the log
[current.txt](https://github.com/kwindrem/GuiMods/files/9522400/current.txt)

kwindrem, 2022-09-08:
added support for v2.90~28 in GuiMods v6.9

closed 2022-09-08 by kwindrem