kwindrem/GuiMods #62: GuiMods file set incomplete

gainestr, 2022-07-29:
Github v6.0  Stored v6.0  Installed v5.35
Running 2.90~08

GuiMods file sent incomplete

Would like the new additions to DC screens

SetupHelper V4.19 installed
Package Manager installed

Log:
@4000000062e3777a2b704aec GuiMods/setup: --- starting setup script v6.0
@4000000062e3777a32984b1c GuiMods/setup: creating file set for v2.90~8
@4000000062e3777c0eec2314 GuiMods/setup: ERROR v2.90~8 dbus_systemcalc.py no replacement file
@4000000062e3777c2dd3395c GuiMods/setup: ERROR: incomplete file set for v2.90~8 - can't continue

kwindrem, 2022-07-29:
v2.90~8 is a pretty old beta. It would be best if you upgraded to the current release candidate v2.90~14. The incomplete file set issue will be resolved in that version.

If there is some reason you really need to stay with ~8, you can work around the problem by running the following command from the command line:
`touch /data/GuiMods/FileSets/v2.90~8/dbus_systemcalc.py.USE_ORIGINAL`
I can build a special version of GuiMods that includes the v2.90~8 file set if absolutely necessary.

gainestr, 2022-07-30:
Unfortunately, the suggestion above did not work.  Still getting the same error on the same file in the log file.  I will consider upgrading to a new version.

closed 2022-07-31 by gainestr