kwindrem/GuiMods #63: spelling error

Jsrivet, 2022-08-14:
There is an "i" missing in the word devices

[image]

kwindrem, 2022-08-14:
Thanks. I'll fix that with the next release.

Jsrivet, 2022-08-14:
Another question why it says no device when I have devices to connect. I get an error in the log.

[image]

kwindrem, 2022-08-14:
Most likely these are due to a dbus call before the related service or a specific parameter is not yet valid.

The first two have nothing to do with my packages.

The one relating to packageManager is part of my SetupHelper package. Without more context for the error (for example what you or the system is doing at the time) it would be difficult for me to locate the cause. It's probably in the GUI somewhere and probably won't be causing an issue. If you notice something out of place in the GUI or elsewhere in the system do provide the related information.

Jsrivet, 2022-08-14:
it is impossible to change the name for the DCsystem in the parameters

[image]

kwindrem, 2022-08-14:
fixed in v6.3 available now