kwindrem/GuiMods #65: After latest update of mods generator is not recognized

djay-hub, 2022-09-06:
Hello,
After latest update generator is not recognized and therefore not working ie. manual start...

kwindrem, 2022-09-06:
It's working here on my test system using a Raspberry PI with Venus OS v2.90~27, GuiMods v6.7 and RpiGpioSetup v3.7.
Please provide a bit more information about your system:
platform (e.g, Cerbo, etc)
firmware version
GuiMods version
Specific issues you are having

djay-hub, 2022-09-07:
Hello Kevin,
Done fresh clean install of VenusOS v 2.89 with clean install of GUIMods and still not working. Before I installed the GUImMod system was working.
[image]
[image]

peterscarsten, 2022-09-07:
Hello,
I've the same issue on my system. In my case it's a Cerbo GX with firmware version 2.89 and GuiMods 6.7

Do you need further information?

Best regards
Carsten

djay-hub, 2022-09-07:
Hello again,...
Update: 
Reinstalled fresh new VenusOS v 2.87 with GUIMods 6.7 the same thing gen not showing up and is not "active"?

kwindrem, 2022-09-07:
I found an issue that prevented bus-generator-starter from running in Venus OS v2.80-v2.89. Fixed in GuiMods v6.8

Thanks for reporting.

GeneratorConnector should not be installed along with GuiMods now that GuiMods contains all the GeneratorConnector functionality. BUT having GeneratorConnector and GuiMods installed does not seem to cause the issues you are seeing. 

The GuiMods setup script was supposed to uninstall GeneratorConnector but there was a bug that is fixed in v6.8. So installing GuiMods will insure GeneratorConnector is not installed. It IS possible to manually install GeneratorConnector after installing GuiMods but as I mentioned above I don't think that will cause any issues. It is still a good idea not to install GeneratorConnector after installing GuiMods 6.x

closed 2022-09-07 by kwindrem

peterscarsten, 2022-09-07:
Hello,
after installing the new version v6.8 the problem is still the same. The generator is not recognised.

peterscarsten, 2022-09-07:
Now it works.
Thank you for this great plugin and for the great support