kwindrem/GuiMods #66: Toggle relay on/off on the

pontusr, 2022-09-16:


kwindrem, 2022-09-16:
GuiMods already provides this capability. It's on a separate overview page since the others are full.

Yes, I understand that your specific system may show some available space on the flow overview but that space is reserved for other devices that may not exist in your system.

pontusr, 2022-09-19:
Yes, I'm using the the separate relay overview page and it works great!
I understand the limited space for room but was thinking that the bottom "bar" could be scrolled (automatically or manually) when the space isn't enough to show all temperatures/tanks/relays.

houlster, 2024-04-13:
How do you get the relay overview page to show?

kwindrem, 2024-04-13:
There's a user preference in Settings / Display & Language / GuiMods to turn the relay overview page on and off. Then you swipe to it from other pages.