kwindrem/GuiMods #70: A few PV Inverters doesn't look good

dersch81, 2022-09-21:
Hi, i just installed your mod and i like it! But the PV Inverter view is not good at all if there are a few of them in the system.

Is it possible to just split them?

[image]

kwindrem, 2022-09-28:
This should not be happening. I'll look into it

The tile should display each PV Inverter cleanly, up to what will fit in the available space.

Please provide me with a list of the PV Inverters including names. Also let me know what Venus OS version and GuiMods version you are running.

kwindrem, 2022-09-28:
I had not considered a multi-phase systems the per leg totals are overwriting the individual PV inverter information. The leg powers are shown only if there are PV Inverters on one of the AC inputs or on the AC output, but if there are more than one PV inverter locations active, only a total is shown.

I have eliminated the leg powers in that specific situation and only show a power value for all legs.

The details page accessed by touching the PV Inverter tile's center should provide total and leg-specific power for each device. I'm also adding leg total powers on the details page.

closed 2022-09-28 by kwindrem