kwindrem/GuiMods #72: NMEA2000 temperature sensors data not present

seamaster101, 2022-10-05:
Diid any one  manage to read NMEA 2000 temperature sensors on Venus RPI?
I read NMEa2000 tanks - no problem, but don't see temperature sensors data (the system sees the temperature sensor device). Why that could be?

seamaster101, 2022-10-05:
here is little more information:
Is there a way to read NMEA2000 Temperature sensors on Venus RPI? The sensors I have in mind are capable of outputting :
PGN 130316 Temperature extended range
PGN 130312 Temperature
PGN 127489 Engine Paramenters Dynamic
PGN 127493 Transmission parameters
The most common standard PGN for temperature is PGN 130316. Can you make Venus OS read that please or if it already supports it how do I set it up?

virtuvas, 2022-10-05:
I'm sure Kevin will be here shortly to put things in perspective.
Just wondering HOW you could possibly present all that! OK I have a twin screw motorboat so it's double of everything, but even getting coolant temp (from 127489) gbox oil temp (from 127493) is a bit too much imho!
I'd try getting temps for fridges, outside and cabin if you have them on your N2K bus (I do)
However, you most likely have other displays on your vessel, so some prioritising needs to be done.
After running two summers with VenusOS and Kevin's guimods, I see that with a touch on the screen I really like to check no reds appear at the bottom on fuel/freshwater and black water tanks (mainly the last two tbh!). I wouldn't mind having 2-3 important temps when I'm at anchor but I'd struggle to see the point of presenting engine data on the VenusOS screen.
Sorry, not a dig, just wondering where this could lead us!

Now to be (slightly) on point and from past discussions, it looks like some work needs to be done to get that data from the N2K bus and into venus. When trying to show tank data, I came across the issue that some custom (teensy) devices that were doing the PGN building and pumping on the bus where multidevices, doing all sorts of things together. Venus didn't like that, so had to break that up in a way that one device did ONLY 3 tanks. That works out of the box now.
I guess the first test would be to make sure that a N2K device does ONLY temp before searching how you pick that up and create a suitable entry into venus. I don't think I've tried that before, could do a quick test, doubt if that would be enough though.

Wonder how Ruvvi (sp?) senders work and if we could pick on that and do custom temp data from N2K.

cheers

V.

kwindrem, 2022-10-05:
I display any device that presents a temperature dbus service. If it's not on dbus as a temperature sensor it will not be displayed.

closed 2022-10-05 by kwindrem

seamaster101, 2022-10-10:
Maybe I did not explained well…
The NMEA 2000 sensors are capable of output one of these temperature PGN depending on their configuration. It does not mean that we need to read them all. All we need to read is PGN 130316 Temperature extended range, which can give you temperature reading for many possible points on the boat including engine room, fish tanks, batteries, etc. and many others. 
It is very common on larger boats of 50+ feet to have already multiple NMEA 2000 sensor (not on engine). It will be nice to be able to monitor NMEA 2000 temperature reports. Is it possible to make this PGN readable?
[image]

kwindrem, 2022-10-10:
This is not something I can do within GuiMods. The N2K interface within Venus OS would need to be modified to read those PNGs and publish a temperature service to dbus. Then GuiMods would display the temperature without any modification.

I don't have the tools necessary to modify the N2K interface and last time I checked it was not open source anyway.

Your best bet is to submit an enhancement request to the Victron forum on the modifications sub-page:

https://community.victronenergy.com/spaces/31/index.html

Victron would then need to evaluate the request and prioritize it with other work they are doing.

virtuvas, 2022-10-12:
I'm pretty sure I've asked Matthijs last year about that, wasn't at their priorities back then...