kwindrem/GuiMods #75: V6.11 No longer showing PV Charger information

gainestr, 2022-10-06:
Newest version is no longer showing PV charger information.  Just installed and can not see my PV status even though it is night and there is no output from the solar array.  Maybe this is something new when a device reads zero output and it will show up in the morning but for now I went back to v6.10.  
[image]
[image]

kwindrem, 2022-10-06:
Fixed now in v6.12.

gainestr, 2022-10-06:
Confirmed, DC Coupled overview is now displaying PV charger.  I edited and added a couple more observations/issues above.  Looking good!  Love the new Watt/Kilowatt option.  Thank you.

kwindrem, 2022-10-06:
New issues fixed in v6.13

closed 2022-10-06 by kwindrem