kwindrem/GuiMods #76: tank % and volume  dont fit in space, some rearrangement?

virtuvas, 2022-10-06:
as per the title,

[image]

remote console messes up values' presentation, maybe giving them a bit more space against the empty space for the text string to the left of each cell? 
tbh don't remember how it looks in real screen, it may actually be ok locally (will have to test on next visit to the boat)  which makes it even more confusing.

2.91, guimods 6.11

cheers

V.

kwindrem, 2022-10-06:
v6.15 now allows you to select %, units or both in the bars. I've also allotted more width to the bar where it is available.

Let me know what you think.

virtuvas, 2022-10-07:
[image]

perfect!

thanks Kevin!

V.

closed 2022-10-07 by virtuvas