kwindrem/GuiMods #77: Amps instead of watts display

kittyorianda, 2022-10-15:
Is there a possibility to display the ampere numbers instead of the watt numbers everywhere in the mobile overview page? Could you perhaps offer a possibility in the options, whether one would like to display amperes, or watts prioritized?
[image]

davinsoftware, 2023-09-21:
+1. I love GuiMods but my biggest wish was to replace watts with amps. Watts aren't very useful to me since I always think in amps and only really need to know that number. If this is possible to take as a community contribution I'd be open to working on it with some pointers in the right direction.

kwindrem, 2023-09-23:
Watts is a more consistent number since amps for AC and amps for DC are vastly different for similar power values (10 to 1 for a 12 volt DC and 120 volt AC system). My hope is that people would program the gauge limits and start ignoring the actual numbers whatever they are in favor of a visual indication of power relative to the device's maximum.

BUT, I see many people prefer amps and Victron is working toward providing a user preference for watts or amps as the primary power display number.

To do this right EVERY menu that displays watts and not amps and every overview page would need to test the global watts/amps preference and adjust its display accordingly. GuiMods currently modifies about 80 files. This number could jump to a couple hundred. Limiting the change to just the overview pages would be much more manageable.

I have guidelines for making SetupHelper-compatible packages. Included there is the mechanism for creating and updating file sets which you'd need to set up in order to contribute to the GuiMods project.

davinsoftware, 2023-09-23:
Yes, I definitely understand the appeal of watts from an electrical purist perspective :)  For context I think in the US we mostly think in amps because all our other boat devices have generally been based in that, and don't have power bars to show usage relative to maximum. Ex, when on AC shore power I know we can use up to 30 amps of shore power. And on DC we know the capacity of our batteries in Ah, so monitoring our amp usage is easiest to relate that to how much capacity is remaining. 

For what it's worth I don't use the power bars because I feel they make it too cluttered. It's a great tool, it's just tough to show exactly the numbers that everyone prefers - this ties into my idea from the other thread about maybe we can make a custom templating / skinning mechanism. If the GuiMods tried to allow everyone to set all their preferences through menu options, there would be way too many menu options. 

If there were a "Prefer Amps over Watts" setting though I think it would be sufficient to only have that apply to the display tiles on the overview flow.

kwindrem, 2023-09-23:
I have played with layouts that have NO numbers !!! only the gauges. Cleans up the clutter a lot. In retrospect, I regret adding the volts/amps/frequency info on the flow tiles. Several people at Victron don't like it because the screen becomes a sea of confusing and cluttered mess. For the casual user I firmly believe less is more. I would actually prefer time-based graphs for displaying power levels. That way it's easier to see trends: how fast SOC is falling, how long that AC output power spike was, etc. The only graph so far is for solar production.