kwindrem/GuiMods #78: PV daily totals in detail pane

GeoZBee, 2022-10-18:
It would be handy if the details pane for PV Charger (from Flow panel) provided the total power produced during current/last charge cycle.

kwindrem, 2022-10-19:
It would be fairly easy to display the number shown for each PV Charger's Daily History / Today / Yield. I'll need to cram the columns closer together to make it fit. Does that seem correct? Is that enough info?

GeoZBee, 2022-10-22:
I spent a bit of time staring at that screen and have mentally cycled through several visual options. 

In short, I think cramming the yield info onto each controller’s row/line would certainly work. 
But, I got to thinking that this is somewhat more discrete data, from the point-in-time info provided *by* that grid layout. 
So, I wondered if makes more sense to put it up by the “Total power” line above, except labeled “PV yield” or similar, and giving total yield of the entire PV system, then sub-broken down by controller. 
If we wanted to over-achieve, reporting the PV voltage and watts max for the day would be… useful-ish —but if I’m being honest those last two figures are more occasional-novelty (for me) than practical data points.

kwindrem, 2022-10-22:
It occurs to me that the current day's yield is only of value after the sun has set. Otherwise, it's just an increasing number through out the sunny part of the day. Many home PV installations provide a graph of production vs time, e.g., KW/day. I think such information would be far more useful than a snapshot of production during the day. Although you could probably grow used to what to expect over the course of a day and deduce that the array is "doing better than normal" or "doing worse than normal".

Maybe Victron did the right thing in supplying history data, but rather than showing it as a string of numbers, show it as a graph: production over the past 30 days. Possibly different arrays could be shown as a bar of different color all stacked on top of each other.

PV voltage and current were provided more as a diagnostic tool. If you see zero or reduced power, you can check voltage and current to maybe see why. I would consider giving these up for today's and yesterday's production.

Lets keep the discussion going and see what we can come up with

GeoZBee, 2022-10-25:
Ok, I like *that* idea a lot —although it will mean more work for you :)

I think a combined graph similar to what the VictronConnect app offers in its history section for each controller would be great. 
There is certainly plenty of room at the bottom of that details pane for a small bar graph . 

A 7 day bar graph showing each day’s total yield —with today’s yield (in a different color??) as a growing number-graph until sunset— would work well. 

And while 30 days would be… useful, I think 7 days of production listed there would be plenty (require less processing and real estate) and would allow more room for listing yield from multiple controllers (individually) next to each other,  by day —if we wanted to go through *that* effort.