kwindrem/GuiMods #81: One of four PV Charger names not showing in Remote Console GUI.

rmgalley, 2022-10-31:
Since changing the communications cable from VE.Bus to VE.Can on a Victron solar charge controller its name does not show in the main Remote Console GUI. However double clicking on the orange PV Charger box shows it to be there. Previously it was showing when the VE.bus cable was used. I'm using a Cerbo GX with firmware v2.92, Setup helper v4.29 and GuiMods v6.19. Is there an explanation or fix please? I've tried rebooting the Cerbo and giving the charge controller different names to no avail.

[image]

[image]

kwindrem, 2022-10-31:
Very interesting. I'm using the same dbus parameter in both the tile and the details page.

When I don't find a valid name, 'pv1', 'pv2', etc is displayed for the name in the tile. The dashes must be coming from dbus.

What is this solar charger?

Can you send me a screens shot of the dbus-spy output for the device?

If you switch to the Menu then back to Pages, the values are refreshed. Give that a try to see if it is a timing issue.

rmgalley, 2022-10-31:
In my original post I wrongly said I’d changed the communications cable from VE.Bus to VE.Can. That should have said VE.Direct to VE.Can. I want to use VE.Can because it is available and because all available VE.Direct and USB ports had been in use already.

So, the problem charge controller is ‘MPPT VE.Can 250/85 rev2’ running up-to-date v3.11 firmware.

I’ve switched many times between Menu and Pages. It did not help so I don’t think it is a timing issue.

I was unfamiliar with the term dbus or dbus-spy but learn from Wikipedia it is associated with naming bus connected devices. I also read _‘When a process sets up a connection to a bus, the bus assigns to the connection a special bus name called unique connection name. Bus names of this type are immutable - it is guaranteed they will not change as long as the connection exists_’. I am wondering if I am experiencing this problem because the device had already been connected via VE.Direct and so will not allow another instance even though I tried changing the name.

Either way, the MPPT and Cerbo are located in a nearby outbuilding.  I link to the Cerbo via an Ethernet cable to a Windows PC in the house (although the same problem is present on an Android tablet). Looking at the VictronEnergy/dbus-spy page suggests it is the right tool to investigate further but I am not aware of how I would access or use it.

I am wondering if I were to reset the Cerbo GX to factory defaults (which I’ve done before) there would be a fresh start without any naming conflicts? While this wouldn’t offer the information which would be most helpful to you, if a Cerbo factory reset worked, it would be easier for me and some implied conclusions reached.

I could have a go with dbus-spy but I’d need guidance from you.

Thanks for you quick reply.

kwindrem, 2022-10-31:
After closer inspection of the code, I do see a difference. In the flow overview, I'm extracting the value as ".text" and in the details page I'm using ".value". It could be the CANbus driver isn't populating .text for /CustomName. I'm going to change the code to use .value in all places and see if that makes the name appear. GuiMods v6.20 will be posted shortly.

Background (if you are interested):
Victron uses dbus for interprocess communication within Venus OS. Most notably here is from devices like your charge controller to the GUI. When a device driver creates a dbus "service" those parameters are then available to other processes.

In both places in the GUI, I subscribe to the parameter /CustomName for all services with a service name starting with com.victronenergy.solarcharger. A unique identifier is appended to that hame for uniqueness. For example mine is com.victronenergy.solarcharger.ttyS7. There is really nothing that remains after the device is disconnected or the system is rebooted.

dbus-spy is a command line tool that allows you to look at the dbus services and in some cases make changes. You need ssh access to get to dbus-spy. How to do that is described here:

https://www.victronenergy.com/live/ccgx:root_access

Jsrivet, 2022-10-31:
I have the same behavior on my display since changing from ve.can to ve.bus. I will follow the update and do the test.

rmgalley, 2022-11-01:
Thanks for the link to 'Venus OS: Root Access'. I have dabbled with PUTTY previously but not so very confident. It is reassuring I'm not the only one experiencing this.

I was going to say I'd await v7.0, but I've just seen the edit. I've just installed v7.0 and that has resolved the problem.

Thank you ever so much.