kwindrem/GuiMods #84: Can I change the order that the temp sensors are displayed?

PjSpain, 2022-11-05:
Can I change the order that the temp sensors are displayed? For example, now they are showing: Freezer, Outside, Inside, Fridge and I would like them to be: Outside, Inside, Fridge, Freezer.

Thanks for any help or advice.

Peter.

PjSpain, 2022-11-05:
[
[image]
](url)

kwindrem, 2022-11-05:
Unfortunately the order is based on the order of discovery to Venus OS. There are ways to sort the list but I have not mastered the technique.

PjSpain, 2022-11-05:
Thank @kwindrem, If I try moving the sensors, that I want to display last, further away from the Cerbo, do you think that will help?

At lease until there's a setting available. 

Thanks.

Peter.

kwindrem, 2022-11-06:
Not sure if blue tooth signal strength would have anything to do with discovery order.

PjSpain, 2022-11-07:
> Not sure if blue tooth signal strength would have anything to do with discovery order.

Thanks for the reply @kwindrem . I did try moving them but no difference. So, I moved them to different places and that worked. They kept the same order and I simply changed the names.

Thanks for your help, keep up the great work.

Peter.

Stormwind99, 2023-02-17:
> So, I moved them to different places and that worked.

Can you clarify what you did?

I have a similar issue because I have so many temperature sensors

Thanks,!