kwindrem/GuiMods #85: Integration of "Shelly 1PM" as PV Inverter / AC Load in 2Setup Helper"

Freensman, 2022-11-06:
Thanks a lot for our work of GuiMods and further the "SetupHelper", which is for me and probably many others "no coding people" a big help to be able to use GuidMods.

So my question / request goes in this direction.

Would you be interested  to team up with "vikt0rm" to implement a possibility to use their "Shelly power meter" addon to measure power, either as a producer (PV inverter / water turbine) or as a consumer.

Discussed here: "community.victronenergy.com/questions/127339/shelly-1pm-as-pv-inverter-in-venusos.html"

It would be very helpfull to have vikt0rm / fabian-lauer integration of shelly on the "SetupHelper" to install this functionality with ease.

Let me know what you think.

kwindrem, 2022-11-06:
I'd be glad to help with creating a package to install Shelly via SetupHelper. I have guidelines in a document that's part of the SetupHelper package and can consult with the designers to help create the package.

Freensman, 2022-11-06:
Okee thanks, I will get in contact with the others how they see it / if they are interested. 

Will check the guidelines as well.

Freensman, 2024-07-19:
Hey kwindrem

some time ago….

We just played around with the Victron Orion XS on GuiMods and noticed that the displayed voltage on the alternator tile is the Output voltage (service battery voltage) but the current is the input current. 

As also the start battery voltage is missing compleatly in victron visualization, we ask our self why not display the Orion XS input voltage, so we can see the start battery voltage as well. 

I guess its a small fix / change where the source voltage is defind for the alternator tile.

Myself unfortunately are not that deep into coding. If I would know where to look for, 
I guess I could change myself. 

But I think it make sence for all users to see values from eighter input or output. 
But I think input makes more sence here for the voltage.

Or a short IN: / OUT: in fromt of the values would help as well.

Cheers and thanks for all that work. 

Silvio

￼

 
Silvio Franceschini
Yachttechniker & Bootbauer
 
y-tronics GmbH
Schachenstr. 39d
8633 Wolfhausen
 
ph: +41 78 708 35 53

https://y-tronics.com/
https://tinyhaussolar.ch/

kwindrem, 2024-07-19:
I am displaying what Victron provides for output current and voltage. So I think this is a bug that should be reported to Victron.

GUI code is QML and fairly easy to understand. Look in /opt/victronenergy/gui/qml/OverviewHubEnhanced.qml and search for id: alternatorBox.

Note that GuiMods install will overwrite this file so you'll need to change it each time GuiMods updates.

Freensman, 2024-07-20:
HeyThanks a lot for this. I will have a look at it and will check with victron about the datas / values you mention. cheers SilvioLet‘s make it work... :-)Silvio Franceschiniessential support world widewww.y-tronics.com

Freensman, 2024-07-20:
Hello 

I just went into "OverviewHubEnhanced.qml „ and checked alternatorBox. 

With my knowleadge I would say you read out:

VBusItem { id: alternatorPower1; bind: Utils.path(alternatorPrefix1, "/Dc/0/Power") }
VBusItem { id: alternatorVoltage1; bind: Utils.path(alternatorPrefix1, "/Dc/0/Voltage") }
VBusItem { id: alternatorCurrent1; bind: Utils.path(alternatorPrefix1, "/Dc/0/Current") }

Form the Victron Excel "CCGX-Modbus-TCP-register-list-3.30.xlsx“ (attached)

from row 567,  I find:

￼

but on row 583 would also be available (which in my eyes would be the start battery voltage:

￼

Not sure where your  "/Dc/0/Power“ is comming from, but looks to be correct.

What I can see in the excel they put more alternator options in with Rev 47, not sure when that happened:

￼

I’m on holiday for two weeks and have no Orion XS at hand to test, but will do so when I’m back. 

May this helps you too  or I’m compleatly on the wrong track :-)

Cheers Silvio

￼

mit freundlichen Grüssen

 
Silvio Franceschini
Yachttechniker & Bootbauer
 
y-tronics GmbH
Schachenstr. 39d
8633 Wolfhausen
 
ph: +41 78 708 35 53

https://y-tronics.com/
https://tinyhaussolar.ch/

kwindrem, 2024-07-21:
My intent was to show output power, current and voltage although output voltage would be redundant since it's shown on the battery tile.

There would typically be a significant amount of voltage drop on the input side if the Orion so not a good indicator of the starter battery voltage.

Freensman, 2024-07-21:
I see your point. On the other side, showing twice kind the same voltage confused / made no sense to me. Sure there is a voltage drop if the alternator is running, but also a indication on how much the alternator is loaded / the cable drop may is. For me the most importat point, if the alternator is not runing / the dcdc not charging, I see a quite exact start battery voltage. May a „IN“ voltage would explain. Anyway, was a suggestion, as may ask for start battery voltage anyway. I will try. Thanks for the help and link to the alternator box. Let‘s make it work... :-)Silvio Franceschiniessential support world widewww.y-tronics.com

kwindrem, 2024-07-21:
I actually had someone insist on showing the output voltage so he could calculate the voltage drop between it and the battery. To me, that's something you check once during installation to verify your design.

There is precedent for showing input values since that's what GuiMods does for a PV charger. And I do label it as such in the details page. So what happens in the PV Charger tile is the individual devices report input power, voltage and current but the total power is what is going to the battery.

It would be fairly easy to do this with the alternator tile and it's details page but I'd want to make a user preference since it's been this way since the beginning of GuiMods.