kwindrem/GuiMods #86: The tile for Grid/Generator/Shore has disappeared from the Remote Console/GuiMods interface.

rmgalley, 2022-11-10:
Thank you Kevin for your work with GuiMods and SetupHelper. If I may I have another one for you.

I’ve an issue somewhat similar to the one last week where one of four PV Charger names was not showing in the Remote Console GUI. I now have the situation where the red tile for Grid/Generator/Shore has disappeared from the Remote Console/GuiMods interface. If I uninstall GuiMods the tile is present and can display, in my case, ‘generator’ power.

In the VRM dashboard, the box and information is displayed whether or not GuiMods is installed.

This situation first began after I’d rearranged the MPPTs, SmartShunts etc. to the VE.Direct and USB ports of the Cerbo GX. Also one MPPT is now connected via VE.Can rather than VE.Direct.

I first tried uninstalling GuiMods then SetupHelper followed by rebooting and then reinstalling. That was not successful so I repeated the above followed by performing a factory default reset to start over. This was not successful either. The Grid/Generator/Shore tile is still absent.

I am wondering if the problem is similar to the ".text" versus ".value" which resolved the former issue.

The Cerbo GX is running firmware v2.92 with Setup helper v4.29 and GuiMods v7.0. Any help or a resolution would be much appreciated.

[image]

[image]

kwindrem, 2022-11-10:
First of all, what appears on VRM has nothing at all to do with the local GUI or what shows up on remote console.

The stock flow overview always displays the AC input tile. I had requests to only show the input tile if there was in fact a Multi/Quattro present in the system. The code looks for exactly one Multi/Quattro and displays the input tile. This test could fail if your system has more than one Mulit/Quattro. (A group of Multis set up for 2 or 3 phases or in parallel count as ONE so that would not hide the AC input tile.) Could this be what is happening? If not I can't explain the behavior.

I could change the test to display the AC input tile. The system identifies the Multi/Quattro it will control and get information from. I can use that information instead of counting the number of Multi/Quattro devices in the system.

rmgalley, 2022-11-10:
Thank you for the quick reply.

You are correct in your diagnosis. There are indeed two VE.Bus devices connected. One is the primary MultiPlus-II via the VE.Bus port on the Cerbo GX. The other is a MultiPlus Compact, connected via a Mk3 - USB Interface, to a USB port. I can confirm removing the USB connected MultiPlus Compact restores the tile and I can see data there again.

The MultiPlus Compact is used to supplement the output power from the MultiPlus-II by being switched by it as a ‘Generator’.

I often favour the Remote Console interface because the information is updated in near real time whereas the VRM Dashboard lags by maybe 10 - 15 seconds. However having the MultiPlus Compact connected allows it to be remotely monitored and, using ‘Remote VEConfigure’, configured and updated.

Thank you for clarifying the relationship between the VRM Dashboard and the Remote Console. I added that screenshot more to illustrate what I was expecting to see in the latter.

If it is possible I would very much welcome being able to view the GuiMods AC input tile with my arrangement.

kwindrem, 2022-11-10:
Should be fixed now in GuiMods v7.1

rmgalley, 2022-11-10:
Thank you Kevin. That has fixed it.

closed 2023-03-22 by kwindrem