kwindrem/GuiMods #87: Flow display PV inverter

lennycb, 2022-11-11:
Hello, recently I've also been using a PV inverter on the AC OUT. Is it correct that only the "overall performance" is displayed?

I know that the tile can also display multiple inverters. Then there is no room for the phase performance.

If there is only one inverter, would it be possible to show the details? Quasi like in org. Victron view.

[image]

kwindrem, 2022-11-12:
It looks like my code should be displaying details in a multi-phase system just like the AC Loads and AC Input tiles.
Screen shots of the original PV Inverter tile as well as dbus-spy for the PV inverter service might help figure this out.

lennycb, 2022-11-12:
GuiMods simple
[image]

Victron stock
[image]

lennycb, 2022-11-12:
I haven't done anything with dbus spy yet. Is there a guide?

lennycb, 2022-11-12:
maybe that will help.
looks like there are no values coming from the individual phases:
A Shelly 3EM is read out
here is the module for it. It could well be that the service does not deliver all registers.

https://github.com/fabian-lauer/dbus-shelly-3em-smartmeter

According to the register overview, the register for the "com.victronenergy.pvinverter" service should be:
L1 power 1029
L2 power 1033
L3 power 1037
Total power 1052

In the ioBroker, only "Total Power" is filled by the Modbus adapter.
Is there code built into the tile that hides "null values"? 
[image]

lennycb, 2022-11-12:
I'm a little further.
It is now clear why the values in ioBroker are not displayed or are displayed with 0.
Values from 0-65535 are permitted for the registers.
[image]

Now the values are negative for me, so that the power fed in is also correctly deducted at AC Load.

This results in 2 approaches:
On the one hand, the developer of the Shelly module must/should do it again.
You would also have to look again at GuiMods, as switching to "positive values" didn't result in any display in the tile.

Can I help?

kwindrem, 2022-11-12:
Victron's convention is for all power and current values to be positive for the "normal" flow for the device. That suggests that the Shelly device is not conforming to Victron specs. It should be reporting positive values.

That said, the stock overview is showing values so my code should also be showing them. Let me look into the code.

But, the negative value will impact other things like DC System power and my gauges won't work. I would need to drill down into the service and figure out if it was a Shelly product and adjust things accordingly.

There are other devices that are in the same situation where power flow has a sign opposite to what Victron deals with on dbus. A "driver" layer usually exists and the change could be made there.

dbus-spy is a command line widget that shows dbus services and parameters. There's a brief explanation here:

https://github.com/victronenergy/venus/wiki/commandline-introduction#working-with-d-bus

and more here:

https://github.com/victronenergy/dbus-spy/blob/master/README.md

kwindrem, 2022-11-12:
The stock simple flow overview shows phase details only if there is a PV inverter in only one location (AC input 1 or 2 or the AC output). 

If more than one exists, the stock overview shows only a sum of all inverters and phases. The area reserved for phase power is then blank.

My GuiMods Simple flow page currently does not show any phase details, but does show up power total (all phases combined) for up to 3 PV inverters.

I can add a case for just one PV inverter as is done in the stock flow overview and show phase power values.

What SHOULD be displayed if there is more than on PV multi-phase inverter? Options are: 1) show power totals for up to 3 PV inverters; 2) show phase totals of all inverters? (1) is the current implementation in GuiMods.

The details page does show individual leg power totals for all PV inverters plus total and leg power for all PV inverters in the system.

lennycb, 2022-11-13:
> 

That is exactly my opinion. Values should all be positive and subtracted or added in the appropriate tile.

What my Shelly is doing here is not yet clear to me.
The measuring transformers are connected in the direction of flow K-->L.

If I do this in the supply line to the house distribution, Shelly shows positive performance.
If I do that on the line from the WR to the house distribution (WR-->K-->L-->House) then I get negative values.

Whatever the case, if you turn the measuring transformer around, it's already positive, but the "AC loads" display no longer fits.

Maybe I have a misconception:
Question:
If I have 2000W at the AC OUT and the WR generates 2000W at the same time at the AC OUT, 0W should be displayed. Correctly?
For me it is added to 4000W.

I had assumed that Victron would handle this correctly since I assigned the Shelley to the PV Inverter role.

Is now not a problem of GuiMods, but maybe you know the problem.

lennycb, 2022-11-13:
##############
What SHOULD be displayed if there is more than on PV multi-phase inverter? Options are: 1) show power totals for up to 3 PV inverters; 2) show phase totals of all inverters? (1) is the current implementation in GuiMods.
##############

So it would be really cool:

- in the case of an inverter, the phase sum at the top and the individual phase power at the bottom
- if there is more than one WR above, the sum of all WRs and below the phase sum of the individual WRs (I think that's the way it is)

kwindrem, 2022-11-13:
If you have a 2000 watt load and your PV inverter is producing 2000 watts then you should see 0 watts for the AC input (assuming no charging going on). I believe the AC Load tile should continue to display 2000 watts. GuiMods also computes an inverter power and that should be 0 as well.

Where are you seeing 4000 watts? If it's showing on the AC output tile then maybe the system is not recognizing the PV inverter being connected to the AC output.

There is probably a "driver" layer that creates the dbus service for the PV Inverter. That's probably where the sign change needs to occur. It may need to be based on the device type to avoid issue with other PV inverters.

kwindrem, 2022-11-13:
Currently, the stock PV inverter tile displays a single power total if there are more than one PV inverter in the system.

The question is really: Which is more important: totals for a single PV inverter or phase totals for all PV inverters.

The details page shows total power for each individual inverter as well as the phase powers.

lennycb, 2022-11-13:
I will observe and calculate with the "AC-Load" tile.

To the PV tile:
It's not really important.
But if you only have one WR, the field looks quite strange (so big and empty).
Hence the consideration to display some more details with only one WR.

What do you mean?

kwindrem, 2022-11-13:
If you have more than one multi-phase PV inverter, is it more important to display the total power for each PV inverter or to display the total power for each leg?

For a single multi-phase PV inverter I can fill the space with individual leg powers. So the tile would show total PV inverter power plus L1, L2 and L3 powers like the stock PV Inverter tile.

For a single single-phase PV inverter the tile would still be mostly empty showing only the total power for that single PV inverter.

For multiple single-phase PV inverters, I'd continue to show total power for all PV inverters plus individual lines for each PV inverter showing it's name and its power.

lennycb, 2022-11-14:
###################
For a single multi-phase PV inverter I can fill the space with individual leg powers. So the tile would show total PV inverter power plus L1, L2 and L3 powers like the stock PV Inverter tile.

For a single single-phase PV inverter the tile would still be mostly empty showing only the total power for that single PV inverter.

For multiple single-phase PV inverters, I'd continue to show total power for all PV inverters plus individual lines for each PV inverter showing it's name and its power.
###################
that's exactly how it would be good.

Saharel001, 2023-03-02:
Hi, any news about this? I have the same situation an it would be help to see L1, L2 and L3 Power in the tile.
[image]

kwindrem, 2023-03-02:
I think this dropped off my radar. I'll look into it today.

kwindrem, 2023-03-03:
A cut at showing phase info on the PV Inverter tile is part of v7.17~4.
To install this version, change the GitHub branch/tag to "beta".
If you don't have auto downloads turned on, tap on Download. If you don't have auto install turned on, tap on Install.
It's in Settings / Package manager / Active packages, then tap on the GuiMods package.

Please check out this version and let me know if it's OK or if there are any issues. I can't test this here as I do not have any PV inverters.

Saharel001, 2023-03-08:
@kwindrem Hi, i test it today, but in my case its doesn't work. Maybe @lennycb can give feedback too.

lennycb, 2023-03-08:
@Saharel001 @kwindrem unfortunately didn't work for me either.

Saharel001, 2023-03-08:
In the beta version the gauge in the top of the tile is gone too

kwindrem, 2023-03-10:
I just pushed v7.17~9 which might have a fix for the PV Inverter tile "not working". I'm guessing since I have not received any details about what isn't working.

Regarding the gauge, I also added code to get the PV Inverter on input max power if there's no PV Inverter on output so you might have to set that parameter in the Power meters settings.

Saharel001, 2023-03-10:
Sorry for my late reply, i wanted to fix this in a fork of your repo to help but now i can say "you are my hero" ;)
[image]

Saharel001, 2023-03-10:
Ignore my message about the gauge, i dont read your post exactly, it works fine.

Saharel001, 2023-03-14:
> @Saharel001 @kwindrem unfortunately didn't work for me either.

@lennycb please test the new beta

kwindrem, 2023-03-14:
I have merged these changes into "latest". The beta is no longer active so please change the branch/tag back to "latest".

If you are referring to the gauge for the PV Inverter tile, I fixed a bug that was always using the max power for the PV on output even if the inverter was on the input side. I now select the max power for PV on input if there is no PV on output in vertex so you need to set that parameter.

closed 2023-03-14 by kwindrem