kwindrem/GuiMods #88: Display cell-voltage (min/max) and CCL/DCL

WaldemarFech, 2022-11-15:
I would love it to see the max and min cell voltages  or (and?) at least the resulting current CCL and DCL (Charge Current Limit / Discharge Current Limit)

The CCL / DCL would be nice in the Main-View (Battery-Field) directly.
The min/max cell-voltages could stay in detailed-view

andkirby, 2022-12-28:
Hi @kwindrem, 

I'd love to have it as well, and few things more :) Like Charge Current Limit, Charge Voltage Limit, etc.
Probably, I'd extend [the battery file](../blob/main/FileSets/DetailBattery.qml), but I got no idea where you read about the API.
I see there are references to the data, but where it's coming from?

For instance, how did you make [this line](https://github.com/kwindrem/GuiMods/blob/main/FileSets/DetailBattery.qml#L34)?
```python
VBusItem { id: batteryLife; bind: Utils.path(systemPrefix, "/SystemState/BatteryLife") }
```

kwindrem, 2022-12-28:
I'm afraid real estate for additional parameters isn't available.

Things like battery life that are in com.victronenergy.system are likely calculated in dbus-systemcalc service.

Holiday43, 2023-01-06:
> I'm afraid real estate for additional parameters isn't available.
> 
> Things like battery life that are in com.victronenergy.system are likely calculated in dbus-systemcalc service.

Hi @kwindrem,

Would it be possible to display the battery cell voltage from the BMS like the temp or tank level boxes at the bottom of the display?  The BMS constantly displays the Lowest cell v and the highest cell v but it would be much more useful if it was not 3 menus deep on the device list.

Thanks very much for all your efforts.

larsenf, 2023-03-09:
Hello @kwindrem I am also interested in a feature or way to view the min/max cell voltages easier. Also if there is a way to more easily view the remaining ah capacity of the battery that would be great, the ah used in the details screen isnt very useful when compared to the eh remaining in my opinion.

kwindrem, 2023-03-10:
I'd be willing to look into possibilities but will need your help. I don't have a BMS that reports cell voltages etc. So I will need you to test potential changes and report back. I may need screen shots and/or logs and/or excerpts from dbus-spy. If you are up to this then let me know. The first thing I need is a screen shot of what dbus-spy shows for your BMS showing all parameters.

Calculation of remaining AH will be the consumed AH subtracted from capacity and would not take into account discharge inefficiencies. If that isn't good enough I'm afraid I wouldn't be able to make the necessary computation since it relies on discharge rate and it's history.

larsenf, 2023-03-10:
Pages:
[image]
Battery detail:
[image]
Menu (Device list) -> SerialBattery -> Cell Voltages 
[image]
Menu (Device list) -> SerialBattery -> Details
[image]
[image]

So all the info im interested in is already availible, its just several menus deep. I'm willing to help anyway i can, if you need a terminal output just tell me what to do.

kwindrem, 2023-03-11:
I have a beta version out for your review. Change the branch/tag for GuiMods to "beta". I await your comments. Thanks.

larsenf, 2023-03-12:
This is the only difference i see:
[image]

kwindrem, 2023-03-12:
I need to see your dbus-spy dump of the services (first page) and also the parameter list for the battery service.

The GUI log might also be helpful

tail -50 /data/log/gui/current | tai64nlocal

larsenf, 2023-03-12:
Services
com.victronenergy.adc
com.victronenergy.battery.ttyUSB1                                                           SerialBattery(LLT/JBD)
com.victronenergy.fronius
com.victronenergy.generator.startstop0
com.victronenergy.logger
com.victronenergy.modbusclient.tcp
com.victronenergy.packageManager                                                                   Package Manager
com.victronenergy.platform
com.victronenergy.qwacs
com.victronenergy.settings
com.victronenergy.shutdown                                                                         ShutdownMonitor
com.victronenergy.solarcharger.ttyUSB0                                              SmartSolar Charger MPPT 150/35
com.victronenergy.system
com.victronenergy.vebus.ttyUSB2                                                    MultiPlus-II 48/3000/35-50 120V

com.victronenergy.battery.ttyUSB1
Alarms/CellImbalance                                                           -
Alarms/HighCellVoltage                                                         -
Alarms/HighChargeCurrent                                                       0
Alarms/HighChargeTemperature                                                   0
Alarms/HighDischargeCurrent                                                    0
Alarms/HighTemperature                                                         0
Alarms/HighVoltage                                                             0
Alarms/InternalFailure                                                         -
Alarms/LowCellVoltage                                                          -
Alarms/LowChargeTemperature                                                    0
Alarms/LowSoc                                                                  0
Alarms/LowTemperature                                                          0
Alarms/LowVoltage                                                              0
Balances/Cell1                                                                 0
Balances/Cell10                                                                0
Balances/Cell11                                                                0
Balances/Cell12                                                                0
Balances/Cell13                                                                0
Balances/Cell14                                                                0
Balances/Cell15                                                                0
Balances/Cell16                                                                0
Balances/Cell17                                                                0
Balances/Cell18                                                                0
Balances/Cell2                                                                 0
Balances/Cell3                                                                 0
Balances/Cell4                                                                 0
Balances/Cell5                                                                 0
Balances/Cell6                                                                 0
Balances/Cell7                                                                 0
Balances/Cell8                                                                 0
Balances/Cell9                                                                 0
Balancing                                                                      0
Capacity                                                                  259.17
Connected                                                                      1
ConsumedAmphours                                                           10.83
CustomName                                                SerialBattery(LLT/JBD)
Dc/0/Current                                                               -1.57
Dc/0/MidVoltage                                                                -
Dc/0/MidVoltageDeviation                                                       -
Dc/0/Power                                                                -93.27
Dc/0/Temperature                                                           30.95
Dc/0/Voltage                                                               59.42
DeviceInstance                                                                 2
FirmwareVersion                                                           0.14.3
HardwareVersion                                         JBD-AP21S001-L21S-200A-B
Info/BatteryLowVoltage                                                      52.2
Info/MaxChargeCurrent                                                         60
Info/MaxChargeVoltage                                                       63.9
Info/MaxDischargeCurrent                                                      40
InstalledCapacity                                                            270
Io/AllowToCharge                                                               1
Io/AllowToDischarge                                                            1
Mgmt/Connection                                              Serial /dev/ttyUSB1
[... 63 lines cut ...]
2023-03-12 00:55:01.026988500 VeQuickView::mousePressEvent: wakeupEvent
2023-03-12 00:56:25.600072500 ScreenSaver::timer
2023-03-12 00:56:25.600079500 ScreenSaver::screenOff
2023-03-12 00:56:58.190788500 VePlatformVenus::wakeupEvent
2023-03-12 00:56:58.190794500 ScreenSaver::enable
[... cut ...]
2023-03-12 01:39:10.642567500 service  "/net/connman/service/wifi_b827eba0a9e4_hidden_none"  was removed but was not found in the services list
2023-03-12 01:39:42.540364500 QVNCServer disconnected from "127.0.0.1"
2023-03-12 01:39:42.540519500 QVNCServer new incoming connection from "127.0.0.1"
2023-03-12 01:39:42.812915500 QVNCServer Client protocol version RFB 003.003
2023-03-12 01:39:42.812922500
2023-03-12 01:39:42.813111500 QVNCServer No password configured, accepting connection
root@raspberrypi2:~#

kwindrem, 2023-03-12:
Made a change. See if that works.

larsenf, 2023-03-12:
Yes! However thousands of a volt detail on the cell voltage would be ideal
[image]

kwindrem, 2023-03-12:
I've increased the voltage precision as you suggested.

Also added max charge/discharge currents. These fields change to "disabled" if the value is zero.

larsenf, 2023-03-12:
[image]
[image]

kwindrem, 2023-03-13:
Another fix to try

larsenf, 2023-03-13:
looks good to me, amazing work!
[image]

kwindrem, 2023-03-13:
incorporated in v7.18.

Please reset the branch/tag to "latest".

larsenf, 2023-03-17:
Hello, I migrated to latest, when will this be implemented?

larsenf, 2023-03-17:
Hi, this never seemed to get incorporated, on 7.18 or the 7.19 I'm on now

kwindrem, 2023-03-17:
You are correct. Apparently my beta branch didn't get merged into main. I'll work on this ASAP.

kwindrem, 2023-03-17:
I just pushed v7.20 to hopefully fix the bad merge. Please check to see if it's working. Thanks.

larsenf, 2023-03-18:
looks like we lost the values for min/max discharge limit
[image]

kwindrem, 2023-03-18:
OK I think it's fixed again

Sorry for the issues. The merge from v7.18 beta branch to main got messed up somehow and all my changes were lost including my file backups.

GuiMods v7.21 should have all the fixes

closed 2023-03-18 by kwindrem

andkirby, 2023-03-19:
Awesome job guys! It's great to have it! 
Cheers