kwindrem/GuiMods #90: Field % from Alternator in the Alternator detail screen

svalbard2, 2022-11-26:
Hi, is it possible to add the Field % on the Alternator detail screen. Now I see you have added the RPM and Temperature which is perfect. Field% would be even better.

kwindrem, 2022-11-26:
There's only so much space in the table so something would need to go.
As an alternative would are to set the gauge limits to match the alternator output and use that for your indication of how much generator power/current is being produced.

svalbard2, 2022-11-26:
OK, I thought maybe it was possible to add one more column, but then no? OK - not so important. I have regulated down the field of mine alternator to prevent it to produce too much and burn so it would be nice to see that the field is all the time below my limitations. The temperature could maybe be on the main screen? Inside the Alernator tile since this is a very important value to follow?

[image]

davinsoftware, 2023-09-21:
+1 for ideas on improvements to alternator tile. 

On the details screen, you could remove RPM and instead show Field %.  RPM is the least useful number since it just tells us the alternator pulley rpm, which is derivable from engine rpm anyway. 

On the overview tile it would be great to have alternator state (float, bulk, etc) and temperature if at all possible. You could eliminate voltage, since that will usually be approximately duplicative of the battery monitor voltage. And I'm not sure why we have a "Total Power" and a "Power"?  I guess for systems with multiple alternators?  On a typical system with one alternator (most common on boats) it's a wasted space on the tile. 

Suggested: 
Power (W or amps)
Temp
Field %

[image]
[image]
[image]

kwindrem, 2023-09-23:
The power total in the detail does support multiple alternators which is apparently quite common in larger boats with two engines.

Switching RPM for Field would require a user preference since I've had the RPM in GuiMods for some time now and I'd most likely get criticized for removing RPM.

Tiles in the DC and AC coupled flow pages typically only display a gauge and a single power number. The Alternator tile displayed in your overview page is special since it replaces the AC input tile in off grid systems that substitute an alternator-like device for an AC generator. I hesitate to add charging state to the Alternator. Firstly, there could be more than one alternator. Secondly, someone will ask for charging state for other DC charging tiles. This would require these tiles to expand in size and there simply is not room.

davinsoftware, 2023-09-23:
Thanks for the info! I agree there is no easy solution that satisfies everyone.  Would it be feasible to allow users to create their own custom screens? Like how CSS stylesheets can be easily overridden by an additional custom file. 

I was thinking GuiMods was based on the HTML5 support in Venus, but I see it's actually Qt files, which wouldn't be simple to template. If it were html files, perhaps there could be a mechanism where users could copy a template HTML5 file, make a few tweaks, upload to Cerbo, and the GuiMods checks for it before loading its default files. I don't know enough about GuiMods design / architecture. Perhaps you have some better idea though.

Obviously custom layouts would only be a power user kind of thing, but it seems like many GuiMods users already have some software experience anyway (since it's somewhat non-trivial for non-technical folks to install).

kwindrem, 2023-09-23:
I have given this some thought over the years and providing a platform for customization does not fit will in the framework Victron has set up. Qt isn't all that difficult but it is yet another language and not something most are familiar with. GuiMods replaces some of the existing .qml files to make GUI changes. It doesn't really change the environment/platform.

The "marine" app is an HTML5 implementation but it does not contain all of the menus present in the main GUI. But with recent changes the app is really starting to be quite useful for user-level system control.

There have been a few users that have modified the Victron GUI or modified GuiMods for their purposes.

I would not want to maintain custom layouts within the GuiMods package. It's already fairly time consuming to keep up with Venus OS changes.

A new GUI currently in development at Victron is still Qt based, but will run on the display computer and allow different views. It uses "web assembly" which must be compiled after all modifications have been made. I still have not figured out how to make this work for GuiMods and my other packages that change the GUI. I can't really say more about the new GUI yet.