kwindrem/GuiMods #91: Generator on/off visually in the GUI?

svalbard2, 2022-11-26:
Hi, I just fixed remote and manual auto start of my generator through the Cerbo GX and a couple of extra relays. 

The problem is that the GUI does not follow the real on/off status of the generator. If I turn the generator on by using the manual switch onboard the GUI still showes that status is OFF and I have the option to start it. Not a big problem, but since the system sences the generator power this should then turn everything ON all over. Is this possible to fix?

kwindrem, 2022-11-26:
Are you referring to the Running/Stopped indication upper left on the enhanced Generator Overview? That indication is not based on AC presence on the generator input. It is based on a digital input which you need to connect to remote indication of generator running.

You can connect this input to your generator control system or a relay on the generator AC output.

The "Detect generator at AC input" item in generator start/stop settings only generates an alarm when power isn't present. It does not hook into any of the generator logic including the Running/Stopped indicator in the overview.

svalbard2, 2022-11-26:
OK, thanks. I am not familiar with the digital inputs on the Cerbo. My ONAN 5500 generator has no real control system either. Just 3 gauges and an ON/OFF+HEAT switch. I mounted two external relays between the Cerbo GX Relay #1 and the ON/OFF switch of the generator to make the auto/remote start to work.

I understand i need one cable with max 5V to signal to the Cerbo and set the digital input port# to Generator.

The Quattro 24/5000 which I have has 3 programmable relays and 2 digital inputs/outputs. Do you think it is possible to make a direct connection between the Quattro and a digital input port of the Cerbo GX? And do some programming of the Quattro when it receives power on the Generator AC IN (AC2 for me) it sends a digital signal out directly to the Cerbo? Then I just need 40cm of a thin cable and done?

kwindrem, 2022-11-26:
You can just put a relay coil on the generator AC output and connect it's contacts to the "generator is running" digital input on Cerbo.

If you can program Quattro to provide a generator presence indication regardless of which input is currently in use then you can probably program it to provide the necessary "generator is running" indication

svalbard2, 2022-11-26:
Thanks, it works fine now. I connected a two-cable on the Digital Input no 1 on the Cerbo and set it to Generator in the Settings - I/O - Digital Input 1. The two other ends I connected to the Programmable Relay K2 (just because that was the easiest to get to behind the two thick 95mm2 positive battery cables) NC and COM.

Then I used VEConfigure for the Quattro, Added two assistants, 
[image]

[image] (SET RELAY OFF)

[image] (SET RELAY ON)

A bit confusing at the beginning until I realised I had to make two assistants, one to open and one to close the relay. Also not intuitive if the two assistants should be ON or OFF. It did not make sence with OPEN and CLOSE but I finally got it to work by trial and error. My first assistant turn the relay OFF (in the Victron terminology). The other ON.

Thanks for your help, a bit complicated wiring just to get the generator running and the GUI to show the correct status, but OK now.