kwindrem/GuiMods #92: Small suggestion - why not move DC to the right?

svalbard2, 2022-11-26:
Hi, I only have DC Loads so I renamed DC System to DC Loads. For input I have the Shore, Generator and an Alternator. 

[image]

Is it possible to move the DC Systemc, or on my screen DC Loads, to the right of the battery? Under AC Loads?

kwindrem, 2022-11-26:
I will look into moving DC System to the right hand column.

While it would technically be possible to dynamically move tiles around (or based on user preferences), it adds a significant amount of complexity to an already complicated layout.

rspott, 2023-01-19:
> [image]

@svalbard2 
Is this picture simply a mockup or do you have this working now?

kwindrem, 2023-03-07:
I have swapped PV Charger and DC System tiles on the DC and AC coupled flow overviews. This is currently in beta test along with other enhancements/changes.

If you would like to test this, go to Settings / Package manager / Active packages and tap on GuiMods. Change the branch/tag to "beta". Don't forget to change this back to "latest".

svalbard2, yes this is implemented and has been for some time. Go to Settings / Display & language / GuiMods and select GuiMods DC Coupled under Flow overview. There's also a GuiMods AC Coupled page for those with PV inverters and/or loads on AC input.

kwindrem, 2023-03-11:
done

closed 2023-03-11 by kwindrem